ros/ros_comm #1266: rosnode.get_api_uri() does not return the actual uri

jrgnicho, 2017-12-14:
https://github.com/ros/ros_comm/blob/247459207e20c1da109fc306e58b84d15c4107bd/tools/rosnode/src/rosnode/__init__.py#L85

When my [client library calls](https://github.com/ROS-PSE/arni/blob/3ae8c7cd6895f770bd689d8a0410d69a6a7afa70/arni_nodeinterface/src/arni_nodeinterface/host_statistics_handler.py#L305) this function I get the following list:
```[1, 'node api', 'my computer name']```

what I'm expecting to get is the host machine's uri; usually the loopback ip 127.0.0.1 if the **ROS_IP** system variable has not been set.
I'm using ros-kinetic on ubuntu 16.04

mikepurvis, 2017-12-14:
If neither `ROS_HOSTNAME` or `ROS_IP` is set, it defaults to using your hostname. That's why roslaunch includes a self-ping check on startup to make sure that the hostname really does resolve back to itself.

In a network environment with functioning DNS, this configuration is a better default for enabling multi-machine ROS comms than falling back to `localhost` or `127.0.0.1`.

mikepurvis, 2017-12-14:
If you crave gory details, here's the logic:

https://github.com/ros/ros_comm/blob/247459207e20c1da109fc306e58b84d15c4107bd/tools/rosgraph/src/rosgraph/network.py#L106-L161

jrgnicho, 2017-12-14:
@mikepurvis  I see, it makes sense to use the hostname as it is more likely to be different than other machines in the same network.  Thanks for that explanation.

mikepurvis, 2017-12-14:
No problem at all.

closed 2017-12-14 by mikepurvis