ros/ros_comm #1270: Rule for accepting patches which break ABI compatibility

dirk-thomas, 2017-12-19:
Since in the recent past more and more PRs have been discussed (and some of them merged) which do break ABI compatibility I would like to establish a rule (or a set of rules) which we follow in the future in order to avoid having the same discussion on every patch. The possible range goes from:

* most conservative: any patch which breaks ABI will only be considered for future ROS distros and will not be merged into already released ROS distros
* <anything in between>
* no restriction: patches which break ABI can be merged into the latest already released distro, probably even be considered for backporting into e.g. an LTS ROS distro.

What I certainly want to avoid is a rule like "it can be done but it depends on <some undefined criteria>". @ros/ros_comm-maintainers @ros/ros_team Please comment with your opinion on this and describe the criteria when a patch should be considered as specific / detailed as possible.

evenator, 2017-12-19:
What may help is to move away from the current branching model. In many ROS repositories, the default development branch is `DISTRO-devel`, where `DISTRO` is a ROS distribution that has already been released. This discourages active development because any merge to the default development branch is a merge to an existing release. Every PR is a referendum on ABI breakage or behavior changes, so many PRs sit in limbo until the next release.

Here's my proposal:

- Add a `master` branch for current development, and make this branch the default. Users who want bleeding edge source can pull `master`.
- For each new ROS release (i.e. once a year), branch master into a release branch.
- Only bloom from release branches.
- Periodically backport patches (or the entirety of `master`) to the release branch(es).

This model (which is basically git-flow) would alleviate some of the ABI headaches on the front end (PRs) by deferring the decision until backport time. That allows for stricter ABI rules because users who really want the new features can choose to build master. This change would also bring the ROS branching model in line with well-known git branching models. It also solves the common problem where a repository had a `jade-devel` branch that's been released to jade, kinetic, and lunar.

dirk-thomas, 2017-12-19:
Thanks @evenator for the feedback. You mentioned "stricter ABI rules". Can you please clarify what exact criteria you would suggest to apply for such backports.

evenator, 2017-12-19:
I don't have a specific rule in mind, but I think you could get away with a hard-and-fast rule you proposed above:

> any patch which breaks ABI will only be considered for future ROS distros and will not be merged into already released ROS distros

IMO, ABI changes are a clear line in the sand. Behavior changes that don't break ABI are more subjective.

evenator, 2017-12-19:
As an addendum to the above, I think that *most* changes you would absolutely *need* to backport (security patches, for instance) could be implemented without changing ABI. There may be things I'm not thinking of.

pjreed, 2017-12-19:
I like @evenator's proposed branching strategy, although as written, I suspect that will lead to an issue that I'm not very fond of, which is having to merge the same changes into multiple, nearly-identical branches.

For example, let's say that your package broke ABI compatibility from Indigo to Jade, but maintained ABI compatibility across Jade, Kinetic, and Lunar.  As it is right now, your master branch is likely named `jade-devel`, and you make changes to that, then release that branch into Jade, Kinetic, and Lunar, then backport that change (if possible) into `indigo-devel` and release that.

With the proposed strategy, you'd instead make all of your changes in `master` and have separate `jade-devel`, `kinetic-devel`, and `lunar-devel` branches, all of which may be identical, and for every change in master you'd need to process three PRs in order to get it into your three other branches so they can be released.  It's easy to do, but it's also a lot of tedium.

Unfortunately, I don't have a good solution for that off the top of my head.  The "standard" way of handling this would be to name your maintenance branches after the release they're modifying rather than the platform they're targeting; so, for example, If your `master` branch is on 3.x but you need to make some changes to 2.1, you'd create a branch named something like `2.1-patch`, make changes to that, and then release it.  The problem with doing that is since that everything is coupled with a particular version of ROS, even if your package's ABI doesn't change, ROS's may have, and so you may need to have separate `2.1-jade`, `2.1-kinetic`, etc. branches.

mikepurvis, 2017-12-20:
Having a dedicated `master` branch would also be very nice for those of us who are managing our own release infrastructure and/or doing all source builds. That said, which branch would be listed in the `source` stanza of the distribution yamls? It would make sense to list the release branch, but then there's nothing tracking the source branch (for example, for the purposes of doing test/PR jobs).

For a live example, I know that @ayrton04 used the separate master branch / git-flow model in developing robot_localization for the first little while. It looks like he's given up on it now, but I do remember looking through the commit/merge history at one point during that time and being completely overwhelming by the snarl of merge commits everywhere.

As far as rules go, my inclination would be be to disallow ABI breakages in the _APIs documented on wiki.ros.org_. However, because of pimpl and smart pointer usage, many of the internal ROS classes _are_ able to change in size without impacting user code. I would argue (as I did for #1217) that if it can be demonstrated that breaking ABI for an internal class doesn't actually impact the documented ABI, it can still be considered.

Another possibility to consider would be having different rules for LTS vs. non-LTS releases. For example something like:

- The current release may always break ABI (eg, treated as in development/master branch).
- An LTS release which is not the most current one may not break ABI.
- Users who want stable ABI should remain on the old LTS until the new LTS is one year old, after which point it will be stable and they can migrate.

dirk-thomas, 2018-05-22:
First I am sorry to the long silence - time for ROS 1 maintenance is very rare these days. I appreciate your feedback and will try to comment / summarize the points brought up.

> What may help is to move away from the current branching model.

I just want to clarify that this discussion is not about branch strategies. Independently on how the branching is managed the questions is what kind of changes can go into which distro. So I will simply not comment on the branching aspect here. There is an conversation on discourse about this topic if you are interested in that: https://discourse.ros.org/t/on-git-branching-strategies/4785

> ... absolutely need to backport (security patches, for instance) ...

That is pretty much how a Linux distribution would "judge" proposed changes. If it isn't a security patch or fixing a severe bug it ain't going into any already released distro.

> Having a dedicated master branch would also be very nice for those of us who are managing our own release infrastructure and/or doing all source builds. That said, which branch would be listed in [...]

The now proposed [rolling](https://discourse.ros.org/t/proposed-changes-to-the-ros-releases/4736/) distribution tries to capture this idea.

> disallow ABI breakages in the APIs documented on wiki.ros.org

While the general idea is appealing I have concerns about the applicability of such a rule. First of all the list of documented API would not be easy to collect / check throughout the wiki. Second if a user has read a header from a dependency and uses the exposed API (assuming it doesn't have a glaring docblock "not public" / "don't use") I wouldn't consider breaking that be would be valid. Imo there is simply too much API in ROS which isn't explicitly documented but still kind of stable.

> The current release may always break ABI
> Users who want stable ABI should remain on the old LTS until the new LTS is one year old, after which point it will be stable and they can migrate.

In the currently used release process I don't think this is viable. This would also need consensus on a broader scale. This decision can't be made just for `ros_comm` if other packages in the latest release assume and expect stability.

The above mentioned rolling release proposal would probably following that idea though. While it is rolling it can break ABI / API. And for the next LTS release the ABI / API will be locked in again.

---

Thank you all for your feedback. I will close this ticket for now but feel free to continue commenting.

closed 2018-05-22 by dirk-thomas