ros/ros_comm #1272: Impossible to accelerating ROS time (simulation to faster than real time)

shardator, 2017-12-20:
Hi,

It's hard to believe, but according to my research I am the only one who would like to accelerate simulation of a ROS robot... and it seems to be an almost impossible task.

It looks like when I create a node that generates faster than real time `clock` topic, it totally messes up the simulation. Not even bags can be replayed faster, than real time.

The reason, as I found is that a huge part of the ROS system relies on `WallTime` and `WallDuration`s, and not just by using these classes, but by doing waits and sleeps in wall times relevant in the simulation. 

I tried to create an accelerated version of ROS:

 - Removed the definitions of `WallTime` and `SteadyTime`, replaced them with `typedef`s to `Time`
 - Removed the definition of `WallDuration` and replaced it with `typedef` to `Duration`
 - Scaled all timed waits with my desired acceleration factor
 - Instead of a `clock` topic I share the time from a timer process in shared memory, so all calls to `Time::now()` are in sync

But now I get stuck in various timing issues related to transformations, more specifically I get errors, like:

    Lookup would require extrapolation into the future.  Requested time 1270387817.717999935 but the latest data is at time 1270387817.482000113

(Note that the timestamp is intentionally in the past, as it is an accelerated time, not the real time.)

Now in the stack trace, as I see, this is happening in `waitForTransform()` call, which is called with `Time(0)` by the simulation (stdr_simulator). And this is weird, semantically, as I would think (by the name of it) that `waitForTransform()` is intended to wait, until the transform is available, not throw an exception that it is not (yet).

Doing some more research showed that the problem is somehow related to `BufferCore::getLatestCommonTime()`. Somehow the cached times does not match the cached transforms.

gavanderhoorn, 2017-12-20:
@shardator wrote:
> The reason, as I found is that a huge part of the ROS system relies on `WallTime` and `WallDuration`s, and not just by using these classes, but by doing waits and sleeps in wall times relevant in the simulation.

As I also wrote on the [ROS Answers post](https://answers.ros.org/question/277860), this is a bit vague. I think it would help if you could directly link to an example of some place where `WallTime` is used and it should not be according to you.

Also: you never mention it in this issue, but you're using `stdr` as your simulator. I believe that is important, as it could well be that user code is doing things it's not supposed to.

shardator, 2017-12-20:
Thanks for the guidance. So I will explain.

1. We tried to create a "time accelerated simulation" of our robot. This means that the whole simulation, including all timings should be all accelerated with the same ratio, like 10x. 
2. I created a node that publishes `/clock` topic at an accelerated speed, like sleep 250 usec, increment time by 2500 usec, publish it.
3. It turned out that this works SOMEWHAT well, but it actually does not work the same as real time. Our robot needs to go on a complex, dynamic route and it gets messed all up sometimes.
4. I tried to investigate the issue, and found that ROS timers are actually fired according to `WallTime` and not the simulated time. Since for example sensor data is generated by timers, this means, that in the robot's time frame the sensors are 10 times slower, than in reality.
5. Increasing the sensor frequency SOMEWHAT helped, but it turned out that there are zillions of small other things that do not happen in the same order as in a real-time simulation or in the real world.
6. I continued deeper investigation and found the following crucial places in the code (I'm running latest ROS Kinetic):
   - In `roscpp` module `timer_manager.h`:

         if (!T::isSystemTime())
         {
           timers_cond_.timed_wait(lock, boost::posix_time::milliseconds(1));
         }

   - In `rostime` in `time.cpp` function `ros_wallsleep()`
   - In `steady_timer.cpp` function 

         void TimerManager<SteadyTime, WallDuration, SteadyTimerEvent>::threadFunc()

     in the lines

         boost::chrono::steady_clock::time_point end_tp(boost::chrono::nanoseconds(sleep_end.toNSec()));
         timers_cond_.wait_until(lock, end_tp);
    
      where each timer waits until the calculated fire time, `sleep_end`, but it is given as **wall** time, however the robot should experience accelerated time, so do all internal timers.
7. So I modified ROS core at these places to calculate the real time from the accelerated time taking into account the desired acceleration, and forget about WALL time, which is not something we want to have. More precisely, _we'd like to accelerate walltime too!_
8. But this did not help either, as from this point all kinds of random effects started to happen. Thus I decided to totally synchronize the accelerated clocks of all processes, and instead of the `/clock` topic, started to send the time in shared memory, and in the `Time` class, totally ignore `setNow()` calls, instead read out the SHM time directly in `Time::now()`.
9. To get rid of `WallTime` and `SteadyTime` (as they are not accelerated), I recompiled ROS with redefining these simply as `Time`.

But unfortunately this does not help. In `tf2`, we always end up missing some transformations for some time, and instead of a simulation we see a huge pile of error messages bragging about extrapolation into non-existent times...

The point is, it looks like ROS was not designed with the intent of accelerated simulations, only real time ones.

dirk-thomas, 2018-02-03:
Of course when changing the timing "zillions of small other things do not happen in the same order". And when speeding the time up you might also reach limits where certain callbacks take too long for the targeted cycle time. In general I think that ROS users have been able to use the system (at least with Gazebo afaik) with faster / slower than realtime. So I don't think there is a general problem here.

Since you didn't mention it explicitly in the ticket did you actually switch to [use_simtime](http://wiki.ros.org/Clock#Using_Simulation_Time_from_the_.2BAC8-clock_Topic)?

dirk-thomas, 2018-02-26:
I am closing this for now due to no response. Please feel free to comment and if the necessary information is provided this can be reopened.

closed 2018-02-26 by dirk-thomas

ywangSAIC, 2018-06-18:
I have same issue here. I have created a separate node for publishing "/clock" on a 10X the wall time (use_sim_time has been set). But all other things went wrong as long as using the clock, even at 1X speed. 
Here is what I have seen:
_____________________
without use_sim_time and /clock (real time):
_______________________
| topic     |      rate  | min_delta |  max_delta |   std_dev |   window |
|---------|--------|-----------|------------|----------|----------|
|/A           |     9.996 |   0.09521 |    0.1071   |   0.001542  |  351   |
|/B            |    10.0   | 0.08804   |  0.1096   |   0.003201  |  351   |
|/C             |   50.0   | 0.01551   |  0.02395   |  0.0009444 |  351   |
|/D             |   100.0  | 0.005596 |   0.02066   |  0.001265  |  351   |
|/E               |  9.996  | 0.09564   |  0.1068   |   0.001527  |  1755  |
|/F                | 9.996  | 0.09576  |   0.1066 |     0.001535 |   1755  |
|/G               |  50.0   | 0.01529   |  0.02403  |   0.0009744  | 3511 |
______________________
with use_sim_time and /clock on 1X speed (sim time):
__________________
| topic     |      rate  | min_delta |  max_delta |   std_dev |   window |
|---------|--------|-----------|------------|----------|----------|
|/A             |   10.0   | 0.1   |      0.1  |       4.375e-16  | 30    |
|/B              |           10.0 |   0.1   |      0.1     |    4.375e-16 |  30  |  
|/C               |            10.0|    0.1   |      0.1    |     4.375e-16 |  30 |    
|/D   | 20.34  | 0.0    |     0.2      |   0.0593  |    30     |
|/E      |                   10.0   | 0.1  |       0.1     |    4.375e-16 |  30   | 
|/F       |                     10.0   | 0.1     |    0.1    |     4.375e-16 |  30  |  
|/G        |                          10.0  |  0.1   |      0.1    |     4.375e-16  | 60  |

______________________
with use_sim_time and /clock on 2X speed (sim time): same topics rate table as 1X
__________________
______________________
with use_sim_time and /clock on 5X speed (sim time):
__________________
| topic     |      rate  | min_delta |  max_delta |   std_dev |   window |
|---------|--------|-----------|------------|----------|----------|
|/A  |              6.15   | 0.0 |        0.5  |       0.08105  |   124  | 
|/B                    |     10.0   | 0.1      |   0.1     |    2.837e-15 |  124 |  
|/C                     |         7.95  |  0.1  |       0.3  |       0.04916  |   202 |  
|/D |  20.05 |  0.0 |        0.2   |      0.05292  |   202  | 
|/E                   |      5.707  | 0.0      |   0.4        | 0.07588 |    160 |  
|/F   |                         5.707  | 0.0       |  0.4  |       0.0747    |  160  | 
|/G |                                 7.97 |   0.1  |       0.3  |       0.04907  |   400  |

tfoote, 2018-07-06:
@ywangSAIC Please provide a reproducible example of how this isn't working for you and what you'd expect it to do. The best would be a rostest or unit test which demonstrates what you expect and shows how the behavior is differing. Without more context your statistics don't indicate much.

flixr, 2018-07-12:
@ywangSAIC your issues seems different from the original one to me.

You are publishing the clock at 10Hz (with 0.1s increments when running with 1x).
But you should publish with time increments that are at least as small as the shortest duration, so publish the clock with at least 100Hz and 0.01s increments...