ros/ros_comm #1275: Topic_tools  transform wait_for_start

borgcons, 2017-12-21:
I believe there is an error in the script

        parser.add_argument(
            '--wait-for-start', action='store_true',
            help='Wait for input messages.'
        )
...
       input_class, input_topic, self.input_fn = rostopic.get_topic_class(
            args.input, blocking=args.wait_for_start)

the variable names are not the same.

dirk-thomas, 2017-12-21:
Can you please elaborate which variable names are not the same. I don't see which exact part you are referring to.

borgcons, 2017-12-21:
My error. 

It took a little while to find this in the documentation.

Any internal - characters will be converted to _ characters to make sure the string is a valid attribute name.

closed 2017-12-21 by borgcons