ros/ros_comm #1279: rosbag record: Option for TCP_NODELAY

pumaking, 2017-12-29:
When recording large messages such as Odometry, the messages arrive at the bag in clumps. It would be nice if rosbag record had a --tcpnodelay or --udp option to prevent this from happening.

dirk-thomas, 2017-12-29:
The `TCP_NODELAY` option is a publisher setting. But in the case of recording `rosbag` is subscribing to the topics. Therefore it can't enforce / request that option.

Therefore I don't think this feature request for `rosbag` is feasible. Instead whichever node is publishing the odometer messages would need to provide that option.

pumaking, 2017-12-30:
It's definitely an option that can be added to a C++ subscription: https://docs.ros.org/api/roscpp/html/classros_1_1TransportHints.html. AFAIK, there is no way to add transport hints to a C++ publisher.

pumaking, 2017-12-30:
Also, since Python subscribers can also specify TCP_NODELAY (but not UDP), it would be nice if `rostopic hz` had a `--nodelay` option as well.

dirk-thomas, 2017-12-30:
Sorry, you are right. The option can also be requested from the subscriber side.

Please consider to provide a pull request to add such an option to the `rosbag record` command.

pumaking, 2018-01-02:
I've created a pull request here: https://github.com/ros/ros_comm/pull/1295

pumaking, 2018-01-02:
I also created a pull request for adding TCP_NODELAY to rostopic hz and delay here: https://github.com/ros/ros_comm/pull/1296

dirk-thomas, 2018-02-02:
I close this issue in favor of the pending PR #1295 (to only have one open item). Thank you for working on this.

closed 2018-02-02 by dirk-thomas