ros/ros_comm #1280: Provide nicer split times when using rosbag with `--split` and `--duration`

berkowski, 2017-12-29:
The current behavior splits files at intervals of the provided duration, so given a start time of 01:43:23 and a max duration of 1 hour we'll get files like:

- 2017-12-29-01-43-23_0.bag
- 2017-12-29-02-43-23_1.bag
- 2017-12-29-03-43-23_2.bag

We'd like to be able to generate log splits at "nice" times.  Using the same example:

- 2017-12-29-01-43-23_0.bag
- 2017-12-29-02-00-00_1.bag
- 2017-12-29-03-00-00_2.bag

I am working on a large project transitioning an existing system over to ROS-kinetic and we will implement this kind of behavior, as well as providing the ability to set the timestamp format in the filename.  I am quite willing to share the result if there's interest here.

dirk-thomas, 2018-01-29:
With the given start time and duration the behavior described with the first bullets makes total sense to me as-is.

Please feel free to create a pull request adding an option to the code base to switch to your proposed naming schema.

dirk-thomas, 2018-01-29:
With the given start time and duration the behavior described with the first bullets makes total sense to me as-is.

~~Please feel free to create a pull request adding an option to the code base to switch to your proposed naming schema.~~ Just saw the referenced PR, thanks.

dirk-thomas, 2018-02-02:
I close this issue in favor of the pending PR #1286 (to only have one open item). Thank you for working on this.

closed 2018-02-02 by dirk-thomas