ros/ros_comm #1283: xmlrpc_manager tries to reuse closed connections

mgrrx, 2017-12-29:
I found time again to revisit the CLOSE_WAIT. Although most CLOSE_WAIT problems are already fixed, roscpp nodes still have always one port in CLOSE_WAIT. This can be easily reproduced by just starting a roscore. The rosout process then has one port in CLOSE_WAIT.

As already addressed in #371, the xmlrpc server used in rosmaster does not use HTTP/1.1. Activating it still leads to the same behavior as reported back then.
Surprisingly, the xmlrpc_manager in roscpp tries to reuse the connections even if the server replies with HTTP/1.0 and closes the connection. Adding the keep-alive flag to the request doesn't help - the server will always close the connection.

I see three solutions to this problem:
1. Find out why activating HTTP/1.1 in the xmlrpc servers leads to the behavior described in #371 and fix this issue. 
2. Detect that the server responds with HTTP/1.0 and close the connection accordingly.
3. Do not cache any xmlrpc connection in roscpp.

I have a working solution for solution 2 which I'll upload in a bit.

mgrrx, 2018-01-02:
1. Is implemented in #1287
2. Is implemented in #1284
3. Is then obsolete

flixr, 2020-01-12:
I guess this can be closed as with #1287 merged this doesn't happen anymore in melodic.

closed 2020-02-03 by dirk-thomas