ros/ros_comm #1285: Conversion from nSecs to Time doesn't seem to worl

adnanmunawar, 2017-12-30:
Why would the following implementation not produce an output and hang the program execution:
```
rospy.init_node('yo')
t_nSecs = rospy.Time.now().to_nsecs()
t_secs = rospy.Time( 0 , t_nSecs).to_sec()
```
I know that `t_nSecs` in this case is the time converted from secs to nsecs and is not just the fractional part of time in secs ( such as `rospy.Time.now().nsecs`) but shouldn't it logically work and produce some output 

**Using
ROS Indigo
Ubuntu 14.04**

tfoote, 2017-12-30:
You have a typo in your code sample. One should return nano seconds one should return seconds. Thus the results are different by a factor of 10^9.

```
>>> import rospy
>>> rospy.init_node('yo')
>>> t_nSecs = rospy.Time.now().to_nsec()
>>> t_secs = rospy.Time( 0 , t_nSecs).to_sec()
>>> t_nSecs
1514668081878837108
>>> t_secs
1514668081.878837
```

This does not appear to be a bug. If you have follow up questions please as on https://answers.ros.org

closed 2017-12-30 by tfoote

adnanmunawar, 2017-12-31:
I just noticed the typo, my bad in rewriting rather than copy pasting the code.
With the typo removed, the execution still hangs on my system.