ros/ros_comm #1288: Same MD5SUM for Accel and Twist messages

zivsha, 2018-01-01:
I'm not sure which version i used to create my cpp headers for [`Accel`](http://docs.ros.org/api/geometry_msgs/html/msg/Accel.html) and [`Twist`](http://docs.ros.org/api/geometry_msgs/html/msg/Twist.html) messages but after debugging some issue I was having I noticed that both messages have the same MD5Sum, which makes sense as the only difference between the two is the name of the class. 

But this causes the following code always print "Twist" even when the message's connection_info is as follow:

```cpp
/*
    connection_info_	0x000001eb8fa0b070
        id         23
        topic      "/device_0/sensor_0/Pose_1/pose/accel/data"
        datatype   "geometry_msgs/Accel"
        md5sum     "9f195f881246fdfa2798d1d3eebca84a"
        msg_def    "# This expresses acceleration in free space ...
        header     shared_ptr { size=3 } 
*/

if (m.isType<geometry_msgs::Twist>())
{
    std::cout << "Twist" << std::endl;
}
else if (m.isType<geometry_msgs::Accel>())
{
    std::cout << "Accel" << std::endl;
}
else
{
    std::cout << m.getDataType() << std::endl;
}
```
Is this a bug? or expected behavior? if so how can one distinguish between the two messages?

paulbovbel, 2018-01-19:
Does this mean that a subscriber can consume a geometry_msgs::Accel message in a geometry_msgs::Twist callback?

dirk-thomas, 2018-02-02:
> Is this a bug? or expected behavior?

Currently the hash of a message is based on all the fields with their types and field names. In this case the two messages `geometry_msgs/Accel` and `geometry_msgs/Twist` are identical and therefore have the same hash. While this is certainly a problem in your case this is "by design" so kind of expected behavior.

> if so how can one distinguish between the two messages?

For your case where you have to distinguish these two types you will have to look at the `datatype` (as you have already mentioned yourself) and can't rely solely on the hash.

> whether a static subscriber can consume a geometry_msgs::Accel message in a geometry_msgs::Twist callback.

I am not clear when you mean with "static subscriber". If you create a subscriber for a topic of the other type it will likely still receive and handle the message (since the hash is equal). I haven't tried that explicitly but that is what I would assume since I think the matching is based solely on the hash and not using the datatype string.

dirk-thomas, 2018-02-26:
Since I assume my comment has answered your question I will go ahead and close the ticket. Please feel free to continue commenting.

closed 2018-02-26 by dirk-thomas

zivsha, 2018-02-28:
That's ok, just forgot to close the issue. I agree with your answer.
Thanks.