ros/ros_comm #1297: roslaunch does not abort if a required node fails to launch

paol, 2018-01-04:
If a node marked with required="true" cannot be launched, roslaunch should abort the entire launch.

As it stands roslaunch terminates if required nodes exit, but doesn't terminate if they fail to start at all.

dirk-thomas, 2018-01-04:
Please describe the case "cannot be launched" / "fail to start" and consider to provide a reproducible example.

paol, 2018-01-09:
Apology for late reply, I was distracted by other tasks.

"cannot be launched" is just what it says. Examples of common failures: package can't be found because env is not setup properly, node command can't be found because of typo in launch file, node command can't be found because of deployment screwup, node command is not set as executable, etc. I'm sure you get the idea.

Example launchfile:
```
<launch>
	<arg name="videos_dir" default="/home/paol/videos" />

	<node pkg="video_stream_opencv" type="video_stream" name="C35_stream" output="screen"> 
		<remap from="camera" to="C35/image_raw" />
		<remap from="camera_info" to="C35/camera_info" />
		<param name="camera_name" value="C35" />
		<param name="frame_id" type="string" value="C35" />
		<param name="video_stream_provider" type="string" value="$(arg videos_dir)/C35_synced.mp4" />
		<param name="fps" type="int" value="10" />
	</node>

	<node pkg="nosuchpackage" type="nosuchnode" name="fail" required="true" output="screen">
	</node>
</launch>
```

roslaunch complains it can't start "nosuchnode", but merrily brings up the the other node and stays running anyway.

We rely on required="true" to ensure a broken pipeline isn't left running, but due to this bug this can happen anyway if the problem happens at startup.

dirk-thomas, 2018-01-09:
That sounds like a good improvement. Please consider to contribute a pull request for this.