ros/ros_comm #1300: roslaunch-complete does not suggest arguments

nalt, 2018-01-06:
When typing `roslaunch pkg  example.launch <tab>`, autocompletion should suggest the arguments of the launch file. This is implemented via function _roscomplete_launch in rosbash, which calls roslaunch-complete.

Autocompletion does not work for launch files with arguments that have no default. This causes the following exception at rosbash-complete:44:

```
/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/arg_dump.pyc in get_args(roslaunch_files)
     47 def get_args(roslaunch_files):
     48     loader = roslaunch.xmlloader.XmlLoader(resolve_anon=False)
---> 49     config = load_config_default(roslaunch_files, None, loader=loader, verbose=False, assign_machines=False)
     50     return loader.root_context.resolve_dict.get('arg_doc', {})
     51 

/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.pyc in load_config_default(roslaunch_files, port, roslaunch_strs, loader, verbose, assign_machines)
    451             loader.load(f, config, verbose=verbose)
    452         except roslaunch.xmlloader.XmlParseException as e:
--> 453             raise RLException(e)
    454         except roslaunch.loader.LoadException as e:
    455             raise RLException(e)

RLException: [/opt/ros/kinetic/share/urdf_tutorial/launch/display.launch] requires the 'model' arg to be set
``` 

This exception should be thrown when actually launching, but not during tab-completion, when the user actually tries to specify the arguments. I guess there should be a way to suppress some exception.

**Examples**

Failing:
`roslaunch urdf_tutorial display.launch <tab>`
It fails because the argument model has no default, which causes the above exception.

Working:
`roslaunch gazebo_ros range_world.launch <tab>`
It works because all arguments have defaults.

nalt, 2018-01-06:
@dornhege, roslaunch-complete hasn't been modified since your commit in 2013. Maybe the exception handing has changed?

dirk-thomas, 2018-02-02:
I would think that this never worked before. Any pull request to support this use case would be very much appreciated.

PR #1521 (2018-11-20): roslaunch: add an option in XmlLoader to only load arg tags
This fixes #1300. It's off/false by default, but it's used by `arg_dump` to stop `roslaunch-complete` from failing if any arg has no default value.
commit b6b22e83 (2018-11-20): roslaunch: add an option in XmlLoader to only load arg tags (#1521)

This fixes #1300. It's off/false by default, but it's used by `arg_dump`
to stop roslaunch-autocomplete from failing if any arg has no default value.
closed 2018-11-20 by dirk-thomas