ros/ros_comm #1302: Compilation error "impossible constraint in 'asm'", kinetic rosbag 1.12.12

lumpiluk, 2018-01-10:
Hi!
I'm getting the following error when trying to compile rosbag, version "ros_comm-release-release-kinetic-rosbag-1.12.12-0" (as stated in the *.rosinstall file):

```
==> make -j8 -l8 in '/home/lumpiluk/git/pg2017w/ros/cross-compilation/build_isolated/rosbag'
Scanning dependencies of target rosbag
[ 12%] Building CXX object CMakeFiles/rosbag.dir/src/player.cpp.o
[ 25%] Building CXX object CMakeFiles/rosbag.dir/src/time_translator.cpp.o
[ 37%] Building CXX object CMakeFiles/rosbag.dir/src/recorder.cpp.o
In file included from /usr/include/boost/thread.hpp:24:0,
                 from /home/lumpiluk/git/pg2017w/ros/cross-compilation/src/ros_comm/rosbag/src/recorder.cpp:54:
/usr/include/boost/thread/future.hpp:410:77: warning: unused parameter ‘lk’ [-Wunused-parameter]
             future_state::state get_state(boost::unique_lock<boost::mutex>& lk) const
                                                                             ^
In file included from /usr/include/sys/select.h:30:0,
                 from /usr/include/sys/types.h:219,
                 from /usr/include/stdlib.h:314,
                 from /home/lumpiluk/git/pg2017w/ros/cross-compilation/install_isolated/opt/ros/kinetic-cross/include/ros/platform.h:53,
                 from /home/lumpiluk/git/pg2017w/ros/cross-compilation/install_isolated/opt/ros/kinetic-cross/include/ros/time.h:53,
                 from /home/lumpiluk/git/pg2017w/ros/cross-compilation/install_isolated/opt/ros/kinetic-cross/include/ros/ros.h:38,
                 from /home/lumpiluk/git/pg2017w/ros/cross-compilation/src/ros_comm/rosbag/include/rosbag/player.h:48,
                 from /home/lumpiluk/git/pg2017w/ros/cross-compilation/src/ros_comm/rosbag/src/player.cpp:35:
/home/lumpiluk/git/pg2017w/ros/cross-compilation/src/ros_comm/rosbag/src/player.cpp: In member function ‘void rosbag::Player::setupTerminal()’:
/usr/include/bits/select.h:41:17: error: impossible constraint in ‘asm’
      : "memory");           \
                 ^
/usr/include/sys/select.h:93:26: note: in expansion of macro ‘__FD_ZERO’
 #define FD_ZERO(fdsetp)  __FD_ZERO (fdsetp)
                          ^
/home/lumpiluk/git/pg2017w/ros/cross-compilation/src/ros_comm/rosbag/src/player.cpp:647:5: note: in expansion of macro ‘FD_ZERO’
     FD_ZERO(&stdin_fdset_);
     ^
CMakeFiles/rosbag.dir/build.make:62: recipe for target 'CMakeFiles/rosbag.dir/src/player.cpp.o' failed
make[2]: *** [CMakeFiles/rosbag.dir/src/player.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:365: recipe for target 'CMakeFiles/rosbag.dir/all' failed
make[1]: *** [CMakeFiles/rosbag.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'rosbag': 
  Command '['/home/lumpiluk/git/pg2017w/ros/cross-compilation/install_isolated/opt/ros/kinetic-cross/env.sh', 'make', '-j8', '-l8']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/lumpiluk/git/pg2017w/ros/cross-compilation/build_isolated/rosbag && /home/lumpiluk/git/pg2017w/ros/cross-compilation/install_isolated/opt/ros/kinetic-cross/env.sh make -j8 -l8

Command failed, exiting.
Makefile:46: recipe for target 'install_isolated' failed
make: *** [install_isolated] Error 1
```

dirk-thomas, 2018-01-11:
Please provide more information (e.g. about your system and environment) which allows to reproduce the problem.

lumpiluk, 2018-01-11:
I'm running Ubuntu 16.04 and I'm currently trying to debug a cross-compilation solution of ROS dependencies for the Raspberry Pi 3.

As far as I understand the issue so far, the problem seems to be that `select.h` is included from `/usr/include`, rather than from the ARM-compatible include directory. I may have opened this issue prematurely because after editing the rosbag/CMakeLists.txt to output the contents of `catkin_INCLUDE_DIRS`, `BZIP2_INCLUDE_DIR`, `Boost_INCLUDE_DIRS`, and finally `INCLUDE_DIRECTORIES`, it appears that `/usr/include` is erroneously included in `catkin_INCLUDE_DIRS`. So far I haven't been able to figure out how it got there or how I can prevent this from happening. I'm also confused because thanks to another CMake file passed to `catkin_make_isolated` as `-DCMAKE_TOOLCHAIN_FILE`, the desired include directories are actually at the top of the `INCLUDE_DIRECTORIES` list and I would have expected those to be searched first for `select.h`.

I'm still very new to everything CMake and Catkin. Is there still a possibility for a bug in rosbag to be causing this error? I'm starting to think that the problem probably lies elsewhere...

dirk-thomas, 2018-01-11:
Your description sounds like that there is something wrong in the cross-compilation setup.

closed 2025-05-31 by sloretz