ros/ros_comm #1305: Dangling if-else triggers compilation error | rosbag_storage

warp1337, 2018-01-15:
I am currently building ROS Kinetic from source using the install generator
on Gentoo Linux (32 bit) --- actually, all this for the Pepper robot, but that is just FYI.

However, [rosbag_storage](https://github.com/ros/ros_comm/tree/1.12.0) fails to compile due to two dangling if else statements (1.12.0) in 

1st)
https://github.com/ros/ros_comm/blob/1.12.10/tools/rosbag_storage/src/lz4_stream.cpp#L180

2nd)
https://github.com/ros/ros_comm/blob/lunar-devel/tools/rosbag_storage/src/bz2_stream.cpp#L131

GCC is 5.4.0.

mikepurvis, 2018-01-15:
We would welcome a pull request which resolves these issues.

warp1337, 2018-01-15:
Can do. Will take approx. 24 hrs.

warp1337, 2018-01-16:
Pull request issued: https://github.com/ros/ros_comm/pull/1306

mikepurvis, 2018-01-16:
PR should target the latest devel branch. When we're ready for another release, there will be a new tag. In the immediate term, you can build this repo from source (adjust the generated rosinstall yaml accordingly) or just patch in this one change.

Let's take the discussion to #1306.

closed 2018-01-16 by mikepurvis