ros/ros_comm #1308: Synchronizing topics from rosbag file

HWiese1980, 2018-01-17:
I've seen some questions about this over at answers.ros.org but none has an appropriate answer (if at all).

Synchronization of topics. Works fine from the real system and from a bag file. **But** the latter only works until the playback resets, like, is started with `--loop` and restarts. Then the stamps are suddenly way off, like, several seconds to minutes in the past, depending on how long the bag file has been recorded. That somehow seems to derail the internal workings of a synchronizer, be it a true equal time synchronizer or just an approximate time synchronizer. Sometimes it works, most of the time it doesn't. 

I've already tried to implement my own approximate time synchronizer but still it only works every now and then. 

This happens both with `use_sim_time` activated and deactivated and with publishing and not publishing `/clock` from rosbag (i.e. launching it with `--clock`). 

This is a pretty big deal for me as I'm working with hardware that takes some time and effort to get running again every time I start up my ROS stack. So in order to test my sensor data algorithms I've recorded some sensor data and would like to run my algorithms on the recorded bag file instead of the real hardware. This, like I said, works sometimes and sometimes doesn't because not working synchronization. 

Can this be solved anyhow? Some of the related questions at answers.ros.org are years old and there doesn't seem to be much interest in this matter...

Cheers

dirk-thomas, 2018-01-17:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2018-01-17 by dirk-thomas

HWiese1980, 2018-01-18:
Yes, I do. But firstly, I do not see this as a question but more as an issue because it's some unexpected behavior, and secondly the questions on answers.ros.org regarding this topic have remained unanswered for months, if not years.

dirk-thomas, 2018-01-18:
You might want to link your questions here for future readers to find it. Maybe the questions needs more information for others to provide helpful feedback, maybe nobody knows the "perfect" answer.

Anyway the current description is too fuzzy to act as a bug report or feature request for `ros_comm`.

TobiasLangnerFUB, 2018-05-04:
What worked for me: 
Have another subscriber outside of the synchronizer check if the time stamp of a new message is older than the time stamp of the previous message. If so, call synchronizer.init() and the synchronizer works again.