ros/ros_comm #1315: Request for backport #962 to indigo

mikaelarguedas, 2018-01-30:
This currently prevents from building ROS robot metapackage from source as some packages use the same branch for all distros and cannot use the new location of the headers.

A good way to reproduce is to try to build ROS robot using `catkin_make_isolated --merge` or `catkin build`
if fails with: `fatal error: XmlRpcValue.h` errors in the `diagnostic_aggregator` package where the previous location of the headers is [being used](https://github.com/ros/diagnostics/blob/3275b512f07954c359e5ff9b916c8ca1232db88e/diagnostic_aggregator/include/diagnostic_aggregator/analyzer_group.h#L51)
Note: this will require the follow-up PR(s) to be backported as well (e.g. #1261)

dirk-thomas, 2018-02-05:
I am not sure about this issue. Existing code using the "old" include should continue to work. That was the goal of #962 and the follow up #1261. #1261 still needs to be backported to Kinetic to address the regression in troduced in #962. If there are any cases where this doesn't work please provide more information about it.

Packages which are using the same branch for older ROS distros like Indigo simply can't migrate to the "new" include path. I don't think the patches in `ros_comm` should be backported into Indigo. Downstream packages can simply wait until they support only Kinetic and newer and then migrate to the "new" include path. I don't see any benefit in allowing to do that for Indigo but a risk of regressions as we have already seen them with #962 (which is why #1261 was necessary).

mikaelarguedas, 2018-02-05:
I'm not sure what additional information I can provide. I gave steps to reproduce the issue in Lunar, the error message and the include line that is failing. Can you clarify what information I should add to this issue description ?

dirk-thomas, 2018-02-05:
> I gave steps to reproduce the issue in Lunar

I don't see any mention of "Lunar" in the description. I thought this was about Indigo.

Since there has been no change for Indigo in this repo (regarding `xmlrpcpp`) I don't see how it could have broke anything.

E.g. the following doesn't even mention which ROS distro this should be done for (I though Indigo based on the title):

> A good way to reproduce is to try to build ROS robot using `catkin_make_isolated --merge` or `catkin build`

mikaelarguedas, 2018-02-06:
> I don't see any mention of "Lunar" in the description. I thought this was about Indigo.

Indeed it was unclear, I experienced the failure on all active distros except indigo (Kinetic and Lunar).

I was following the "installation from source" instructions that are getting the code via the tar archives and not the upstream-dev branch, so I didn't have #1261 in the code I was compiling. Now that this has been tagged and pushed to the upstream repo for Lunar, it works :+1: 
The problem will be solved on kinetic when #1261 is backported and released so I'm going to close this.

Thanks for the feedback and sorry for the noise

closed 2018-02-06 by mikaelarguedas