ros/ros_comm #1326: branch for Melodic

dirk-thomas, 2018-02-12:
I released a new version of `ros_comm` into Lunar [last week](https://github.com/ros/ros_comm/commit/2e6ac87958b59455aab0442b205163e9a5f43ff2). The same version is also available for Melodic. Since there are [several PRs pending](https://github.com/ros/ros_comm/pulls?q=is%3Apr+is%3Aopen+label%3Aconsider-future-release) which will likely going into a `melodic-devel` branch I would like to branch in the near future.

That will imply that other pending PRs will have to retarget the new default branch (to keep porting patches between branches easy). Are there any specific PRs which should be considered for merging *before* the new branch is created?

@ros/ros_comm-maintainers Please let me know what you think about the timing of branching `melodic-devel`.

mikepurvis, 2018-02-13:
No concerns here— the sooner the better from my point of view.

An additional note: we're going discontinue use of the in-repo "edge branch". Its existence has caused some confusion both internally and externally and every time I regenerate it it spams the included PRs with unnecessary notifications. It is also an issue for the CPR build team as I'm the only one with write access to it. Our workflow is unchanged and we intend to continue testing selected PRs on our devel fleets as we're able, but we'll manage the patching process on our own infrastructure. FYI @tspicer01

dirk-thomas, 2018-04-23:
This has been created by @mikepurvis in the meantime. Thank you!

closed 2018-04-23 by dirk-thomas