ros/ros_comm #1329: Deadlock when using StatisticsLogger

DieHertz, 2018-02-20:
There seems to be a possibility for a deadlock in an otherwise single-threaded application when using `StatisticsLogger` due to how it's publisher is initialized:
```cpp
    if (!pub_.getTopic().length())
    {
      ros::NodeHandle n("~");
      // creating the publisher in the constructor results in a deadlock. so do it here.
      pub_ = n.advertise<rosgraph_msgs::TopicStatistics>("/statistics", 1);
    }
```

Pre-requisites:
1. ros-kinetic, likely present in further versions a well
2. Have multiple subscription to the same topic
3. Enable topic statistics logging

Test sample:
```cpp
int main(int argc, char** argv) {
  ros::init(argc, argv, "statistics_logger_deadlock");
  ros::NodeHandle nh{"~"};

  // This call returns properly
  ros::Subscriber sub_one = nh.subscribe<TopicType>("/topic", 1, callback);

  // In case of a deadlock, the following function never returns
  ros::Subscriber sub_another = nh.subscribe<TopicType("/topic", 1, callback);

  ros::spin();
}
```

After successfully subscribing via `sub_one`, a message arriving to `/topic` via `PollManager` in another thread calls `StatisticsLogger::callback`, which attempts to `advertise` it's own topic.
If, at the same time, in the main thread execution happens to be inside the second `subscribe` call, the two may come into a deadlock because of locking mutexes in the opposite.

Internally `roscpp` handles subsequent subscriptions to the same topic by appending them to the same `Subscription` instance, and that is one of precursors to deadlock problem.
Another is `TopicManager::addSubCallback` calling `sub->addCallback` while `subs_mutex_` is locked.

In my application, the issue manifests itself as follows:
```
(gdb) info thread
info thread
  Id   Target Id         Frame
* 1    Thread 0x7f2fd01f79c0 (LWP 28376) "node" __lll_lock_wait ()
  2    Thread 0x7f2fa13d1700 (LWP 28425) "node" 0x00007f2fc0a7374d in poll ()
  3    Thread 0x7f2fa09be700 (LWP 28493) "node" __lll_lock_wait ()
  4    Thread 0x7f2fa01bd700 (LWP 28495) "node" 0x00007f2fc0a7374d in poll ()
  5    Thread 0x7f2f9f9bc700 (LWP 28496) "node" pthread_cond_wait@@GLIBC_2.3.2 ()
  6    Thread 0x7f2f9f1bb700 (LWP 28836) "node" pthread_cond_timedwait@@GLIBC_2.3.2 ()
  7    Thread 0x7f2f9e97f700 (LWP 29590) "node" pthread_cond_timedwait@@GLIBC_2.3.2 ()
  8    Thread 0x7f2f9e15b700 (LWP 29602) "node" pthread_cond_timedwait@@GLIBC_2.3.2 ()

(gdb) bt
bt
#0  __lll_lock_wait
#1  __GI___pthread_mutex_lock
#2  boost::posix::pthread_mutex_lock
#3  boost::mutex::lock
#4  boost::unique_lock<boost::mutex>::lock
#5  ros::Subscription::addCallback
#6  ros::TopicManager::addSubCallback
#7  ros::TopicManager::subscribe
#8  ros::NodeHandle::subscribe(ros::SubscribeOptions&)
#9  message_filters::Subscriber<...>::subscribe
#10 message_filters::Subscriber<...>::Subscriber
#11 (anonymous namespace)::Subscription::Subscription
#12 std::make_unique<...>
#13 (anonymous namespace)::Actor::Actor
#14 main

(gdb) thread 3
thread 3
[Switching to thread 3 (Thread 0x7f2fa09be700 (LWP 28493))]

(gdb) bt
bt
#0  __lll_lock_wait
#1  __GI___pthread_mutex_lock
#2  boost::posix::pthread_mutex_lock
#3  boost::mutex::lock
#4  boost::unique_lock<boost::mutex>::lock
#5  ros::TopicManager::advertise
#6  ros::NodeHandle::advertise
#7  ros::Publisher ros::NodeHandle::advertise
#8  ros::StatisticsLogger::callback
#9  ros::Subscription::handleMessage
#10 ros::TransportPublisherLink::handleMessage
#11 ros::TransportPublisherLink::onMessage
#12 ros::Connection::readTransport
#13 ros::TransportTCP::socketUpdate
#14 ros::PollSet::update
#15 ros::PollManager::threadFunc
#16 ??
#17 start_thread
#18 clone
```

It may be an option to use fine-grained locking in `TopicManager::subscribe`, subscribing only before pushing to `subscriptions_` and inside `TopicManager::addSubCallback` only while we're searching `subscriptions_`.
It has an adverse effect of allowing to create multiple `Subscription` instances for the same topic when calling `subscribe` from different threads (is that supported to begin with?).

Another solution would be adding another lock for mutually excluding `subscribe` and `advertise`, rendering the whole situation impossible.

We could also move `StatisticsLogger::pub_` initialization to `StatisticsLogger::init`, forcing it to be called from the main thread where the user is subscribing.

I may submit a PR with one of these solutions, or a different one not listed here, after musing about it for a while.

DieHertz, 2018-02-22:
After reflecting on the issue for a while, I've concluded that the last option is the way to go, to not break any contracts.
Options 1 and 2 may change the behaviour of existing code which doesn't use `StatisticsLogger` at all, and introduce performance regressions, while with option 3 changes are limited `StatisticsLogger` only.

I'll submit a PR for discussion and review.

ironmig, 2018-05-23:
Ahhh I finally understand the deadlock. Here's my annotation of the backtrace for future reference.  Thanks for finding this!
```
New subscriber to same topics
---------------------------------------
#4  boost::unique_lock<boost::mutex>::lock
#5  ros::Subscription::addCallback                           <-- locks callback_mutex_
#6  ros::TopicManager::addSubCallback
#7  ros::TopicManager::subscribe                             <-- locks subs_mutex_
#8  ros::NodeHandle::subscribe(ros::SubscribeOptions&)

Previous subscriber to topic receiving first message and therefore advertising /statistics
------------------------------------------------------------------------------------------------------------
#4  boost::unique_lock<boost::mutex>::lock
#5  ros::TopicManager::advertise                     <-- Locks subs_mutex_
#6  ros::NodeHandle::advertise
#7  ros::Publisher ros::NodeHandle::advertise
#8  ros::StatisticsLogger::callback
#9  ros::Subscription::handleMessage                 <-- Locks callbacks_mutex_
```

DieHertz, 2018-05-23:
Added your annotation to issue description.
I'm glad to be of help!

peci1, 2020-10-02:
I've faced a probably related issue in Melodic. I still don't know the deepest culprits, but `/enable_statistics` definitely plays a role there.

I haven't gotten into a deadlock, but right to a segfault in mutex locking code. This is a minimum segfaulting example.

```c++
#include <ros/ros.h>
#include <nav_msgs/Odometry.h>
#include <sensor_msgs/PointCloud2.h>
#include <diagnostic_msgs/DiagnosticArray.h>
#include <boost/thread.hpp>
#include <tf/transform_broadcaster.h>
#include <tf/transform_listener.h>

void pub(tf::TransformBroadcaster& broad)
{ 
        ros::Rate r(10);
        while (ros::ok())
        {
           tf::StampedTransform t;
           broad.sendTransform(t);
           r.sleep();
         }
}

int main(int argc, char** argv)
{
        ros::init(argc, argv, "asd");

        tf::TransformListener ear(ros::Duration(30), true);
        tf::TransformBroadcaster broad;

        ros::NodeHandle n;
        ros::Publisher mapPub = n.advertise<sensor_msgs::PointCloud2>("point_map", 2, true);
        ros::Publisher scanPub = n.advertise<sensor_msgs::PointCloud2>("corrected_scan", 2, true);
        ros::Publisher outlierPub = n.advertise<sensor_msgs::PointCloud2>("outliers", 2, true);
        ros::Publisher odomPub = n.advertise<nav_msgs::Odometry>("icp_odom", 50, true);
        ros::Publisher odomErrorPub = n.advertise<nav_msgs::Odometry>("icp_error_odom", 50, true);
        ros::Publisher diagPub = n.advertise<diagnostic_msgs::DiagnosticArray>("icp_diagnostics", 2, true); // 6th publisher -> death

        boost::thread pubThread = boost::thread(boost::bind(&pub, broad));

        ros::spin();

        pubThread.join();
}
```

The segfault happens on line

https://github.com/ros/ros_comm/blob/121bac963e90b9a0bb33cc368ef4373a2f65ddff/clients/roscpp/src/libros/publication.cpp#L437

Debugger told me that `this` in the place of the segfault, was a null pointer. However I'm not sure if it wasn't some kind of corruted memory instead.

I've recorded a bag file during whose replay this segfaults: https://drive.google.com/file/d/1O_2ELqmsSgLzaWMlWVmQ11jqbuvZBe_L/view?usp=drivesdk. Play with `use_sim_time=true`, `enable_statistics=true`, clock from bag file. As soon as you start this node, it segfaults.

What is weird is that it doesn't segfault if you instantiate only 5 publishers. As soon as you add the 6th, the segfault is there (tested on several x86_64 machines ranging from laptop to server). I tried substituting the publishers with an equally long WallTime sleep, but that didn't trigger the segfault.

Anyways, this code uses a quite common pattern - a node that has both a transform listener and broadcaster. My view into it is that the broadcaster sends transform, locks some internal roscpp lock, the listener receives the message and wants to start a statistics publisher, but probably the publishing internal lock is still locked? However, that wouldn't explain what the 6 publishers do there and why are they required for the segfault to happen. If I start the transform listener after the publishing thread, everything is okay. That would also be a hole in my theory...

closed 2025-05-31 by sloretz