ros/ros_comm #1334: [roslaunch] private names in param behave unexpectedly

CodingPeon, 2018-02-24:
According to documentation about `<param>`
> You can also set private parameter across a group of nodes by using the ~param syntax (see ROS names) in a <param> tag. The declared parameter will be set as a local parameter in the <node> tags that follow that are in the same scope (i.e. group or ns tag).

However setting private param outside `<node>` causes any parameter declared inside `<node>` block to "leak" to sub sequence `<node>` blocks. I'm experiencing this in Jade distro. Here is how to reproduce:
    <?xml version="1.0" encoding="utf-8"?>                          
                                                                    
    <launch>                                                        
                                                                    
        <param name="~num_drones" value="2"/>                       
                     
                                                                    
        <node name="A" pkg="drone_core" type="drone.py">           
            <param name="local_name_leaked" value="leaked" />       
        </node>                                                     
                                                                    
        <node name="B" pkg="drone_simulator" type="sim.py"> 
          <param name="local_name" value="debug"/>                  
        </node>                                                     
                                                                    
    </launch>                                                       

  
With this `<param name="~num_drones" value="2"/>`, I get:

    PARAMETERS                      
     * /A/local_name_leaked: leaked 
     * /A/num_drones: 2             
     * /B/local_name: debug         
     * /B/local_name_leaked: leaked 
     * /B/num_drones: 2                   
     * /rosdistro: jade             
     * /rosversion: 1.11.21         

With this `<param name="num_drones_local" value="2"/>` I get:

    PARAMETERS                       
     * /A/local_name_leaked: leaked  
     * /B/local_name: debug          
     * /num_drones_local: 2          
     * /rosdistro: jade              
     * /rosversion: 1.11.21

cwecht, 2019-03-11:
I was not able to reproduce this using melodic.

dirk-thomas, 2019-03-29:
Based on the comment I will go ahead and close this ticket for now. Please feel free to provide more information how to reproduce the problem and the ticket can be reopened.

closed 2019-03-29 by dirk-thomas