ros/ros_comm #1339: numerical string param is turned into int or float even with quotes around it using args or rosparam set

lucasw, 2018-03-03:
If a numerical string like 4235 or 57.23 is set onto the parameter server as a command line argument to a C++ node then getting that param doesn't work.

This doesn't work:

```
#include <ros/ros.h>

int main(int argc, char** argv)
{
  ros::init(argc, argv, "string_param_test");
  ros::NodeHandle nh("~");
  std::cout << ros::this_node::getNamespace() << "\n";
  std::cout << nh.getNamespace() << "\n";

  // this doesn't work if str_param was passed in as argument, or set
  // rosparam set /string_param_test/str_param 98432
  std::string test = "";
  nh.getParam("str_param", test);
  std::cout << test << "\n";

  // this works, and if this program is run a second time the above
  // will get 12345
  nh.setParam("str_param", "12345");
  nh.getParam("str_param", test);
  std::cout << test << "\n";

  return 0;
}
```

```
$ rosrun param_test string_param_test _str_param:="98432"
/
/string_param_test

12345
```

This works fine:
```
#!/usr/bin/env python

import rospy

rospy.init_node("string_param_test_py")
print rospy.get_param("~str_param", "")
```

```
./string_param_test.py  _str_param:=5
```

It looks like the root of the issue is that setting a parameter with quotes that is a valid int or float turns into an int or float

```
$ rosparam set /string_param_test/str_param '8945'
$ rosparam get /string_param_test/str_param 
8945
```

But setting a parameter explicitly as a string with nh.setParam results in a string:

```
$ rosparam get /string_param_test/str_param 
8945
```

I'm using kinetic on Ubuntu 16.04.  

I also tried this with kinetic-devel, though got a `‘ros_steadytime’ is not a member of ‘ros’` error (even when bringing in the latest roscpp_core) so went back to d4788bc8aa1fc354448d5d7ad45d45451668eade .

(I experimented with adding unit tests here but I'm not sure how to quickly build and test them in a catkin tools build environment - it takes a long time to build and run many tests of test_roscpp, I can't selectively just do a few.  How to build just `./devel/lib/test_roscpp/test_roscpp-params` and not run it or anything else?  Extract the build line out of a verbose build/run of everything?)

lucasw, 2018-03-04:
Double quotes works for rosparam set:

```
rosparam set /test "'98432'"
rosparam get /test 
'98432'
```

But I can't use that as an arg:

```
rosrun param_test string_param_test str_param:="'98435'"
terminate called after throwing an instance of 'ros::InvalidNameException'
  what():  Character ['] is not valid as the first character in Graph Resource Name ['98435'].  Valid characters are a-z, A-Z, / and in some cases ~.
Aborted (core dumped)
```

lucasw, 2018-03-04:
Double quotes doesn't work for roslaunch <param> tags, the inner single quotes become part of the string as with args to a C++ node.

gavanderhoorn, 2018-03-09:
I don't see it in your example invocations, but is `'"12345"'` also problematic? The `'` should prevent the shell from evaluating the argument, so I would expect the `"` to be preserved.

lucasw, 2018-03-09:
`'"12345"'`  results in 12345 of type string when given to rosparam set or rospy arg so my pr makes C++ and roslaunch args match that.

Would the argument for going the other way and making '"12345"' preserve the inner quotes in all cases be that there isn't otherwise a good way to set a string with quotes around it?  

With some further experiments with quotes within strings it looks like there is an additional issue:

```
$ rosparam set /foo "'foo' blah" 
Traceback (most recent call last):
  File "/opt/ros/kinetic/bin/rosparam", line 35, in <module>
    rosparam.yamlmain()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosparam/__init__.py", line 626, in yamlmain
    _rosparam_cmd_set_load(command, argv)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosparam/__init__.py", line 543, in _rosparam_cmd_set_load
    set_param(name, arg2, verbose=options.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosparam/__init__.py", line 375, in set_param
    set_param_raw(param, yaml.load(value), verbose=verbose)
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 71, in load
    return loader.get_single_data()
  File "/usr/lib/python2.7/dist-packages/yaml/constructor.py", line 37, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 39, in get_single_node
    if not self.check_event(StreamEndEvent):
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 174, in parse_document_start
    self.peek_token().start_mark)
yaml.parser.ParserError: expected '<document start>', but found '<scalar>'
  in "<string>", line 1, column 7:
    'foo' blah
```

`rosparam set /foo '"foo" blah'` is the same, but putting quotes later than the first non-space character works:

```
$ rosparam set /foo 'blah "foo"' && rosparam get /foo
blah "foo"
```

So there is some code (in yaml parser not rospy?) that is looking for a leading quote regardless of a matching ending quote and then misinterprets and throws.

gavanderhoorn, 2018-03-09:
Strings in yaml don't necessarily need quotes. That is probably why you don't see them in `rosparam get` output.

lucasw, 2018-03-09:
> Not seeing quotes also doesn't necessarily mean it's not a string.

I've noticed that rosparam prints quotes when it it is ambiguous: `1234` type string prints `'1234'`, but `1234f` is just plain `1234f`.

`-p` pretty print drops the quotes.

(It would be convenient if `-v` printed the type definitively for what is in the rosparam server, I could add that later)

flixr, 2018-08-03:
We are encountering the same problem...

VictorLamoine, 2019-12-17:
For those looking for a work-around:

```cpp
std::string serial_number;
if (!nh->getParam("connect_serial_number", serial_number))
{
  // When using rosrun the serial number is interpreted as an int and
  // can't be forced into a string. See https://github.com/ros/ros_comm/issues/1339
  int serial_number_int;
  if (nh->getParam("connect_serial_number", serial_number_int))
    serial_number = std::to_string(serial_number_int);
}
```

First: try to get the parameter as a string.
If it fails: try to get the parameter as an int and convert it to a string.
With this it's pretty easy to support integers/floats.

flixr, 2019-12-18:
yeah, but that workaround has limitations... e.g. it won't work for strings with leading zeros..

closed 2025-05-31 by sloretz