ros/ros_comm #1343: Regression: rosout node spins rather than blocking and waiting, using 100% CPU

NickZ, 2018-03-05:
Introduced in faab4cbc63d6c750c6dc6d050f6268dd19a2bc68. This is a bug in the roscpp client.

This only occurs if the local Boost library is <1.61. I have tested on both Ubuntu Xenial 16.04 (Boost 1.58) and Ubuntu Bionic 18.04 (Boost 1.65), compiling ros_comm from source.

In `callAvailable()`, if no callback is available, `wait_for()` is called on the condition variable, and is told to wait for a specified amount of time (usually 0.1 seconds). However, instead of waiting for the specified amount of time, it returns immediately, turning this into a spin wait rather than a blocking wait. This leads to the rosout node chewing up 100% of the CPU core.

The reason for this is a problem with the backported Boost condition variable library that is used if the system does not have Boost version >=1.61. I am still trying to figure out what is ultimately the problem, but as near as I can figure out, the timeout variable is not being properly set to a future time.

NickZ, 2018-03-06:
The issue is that, on linux systems and every other platform with monotonic clock set, `wait_for()` calls "steady_clock" to get the system time. "steady_clock" is a monotonic time clock that is a relative clock (usually whenever the system booted up.) However, `wait_for() => wait_until() => do_wait_until()` ultimately calls `pthread_cond_timedwait()` with a timeout value based on that relative clock value. `pthread_cond_timedwait()` only takes in absolute system time values. Thus, it immediately returns with a timeout, since the relative time is guaranteed to be smaller than the system time.

NickZ, 2018-03-06:
Ok, I've discovered the root cause. It's nasty, nasty compiler/linker behavior. Also, the bug is different across boost versions.

`pthread_cond_timedwait()` does not normally take monotonic clock values. However, when creating the condition_variable, it can set the condition attributes of the pthread condition to accept Monotonic clock values: 
[image]

However, this never actually gets called.

**On systems that have boost <1.61 (tested on Ubuntu 16.04)**, it's supposed to use the `condition_variable` defined in boost_161_pthread_condition_variable_fwd.h. However, the constructor for the condition variable as defined in that file does not actually get used, it uses the constructor in the system boost library, which does not call the pthread_cond_init function: 

[image]

This is especially bizarre, since I have observed in debugging that it's using the `wait_for()` and `wait_until()` function calls in boost_161_pthread_condition_variable_fwd.h.

**On systems that have boost >=1.61 (tested on Ubuntu 18.04)**, it uses the system boost library. Normally, the pthread condition attribute would be set, since we set `BOOST_THREAD_HAS_CONDATTR_SET_CLOCK_MONOTONIC` in callback_queue.cpp. However, we can see that's not the case: 
[image]
It uses the preprocessor else block instead. This is even more bizarre, since it hits numerous other code blocks that require that preprocessor definition. 

I don't know how to fix this, unfortunately, but hopefully someone here knows enough compiler/linker juju to fix it.

flixr, 2018-03-06:
> However, wait_for() => wait_until() => do_wait_until() ultimately calls pthread_cond_timedwait() with a timeout value based on that relative clock value. pthread_cond_timedwait() only takes in [...]

That should not really be possible to happen, `wait_for(duration)` always basically calls `wait_until(current_time + duration)` where `current_time` is either derived from system clock or monotonic clock dependent on the `BOOST_THREAD_HAS_CONDATTR_SET_CLOCK_MONOTONIC` define.

I cannot reproduce this issue, could you please provide some more information?

Tried with latest versions from shadow-fixed (ros-kinetic-roscpp 1.12.13 and so on) and the simple listener from ros-kinetic-roscpp-tutorials and it works fine.

kjaget, 2018-03-09:
I saw this same issue building roscpp from source just recently.  Same symptom - near 100% cpu usage in every node. Reverting to 1.12.12 fixed the problem immediately.  I can pull info off my system to help diagnose but I'm not sure what might be useful.  It is a Ubuntu 16.04 system which is up to date with apt-get but with not much custom built - let me know what info would be useful.

NickZ, 2018-03-10:
I did some further testing, and apparently this issue does not occur if you build the entire ros_comm stack from scratch; It only occurs if you either already have the base 1.12.12 system packaging installed and you only build ros_comm and roscpp_core, or if you rebuild only the ros_comm part for a source installation. So, some sort of compiler/linker issue. I have to set up some sort of testing apparatus to track the cause down with more specificity.

flixr, 2018-03-10:
Hm.. strange. I still can not reproduce this...
What I tried on Ubuntu 16.04 with the released kinetic packages (1.12.12) as base:
- create a clean catkin workspace based on /opt/ros/kinetic
- add `ros_comm`, `roscpp_core` and `ros_tutorials` repos with kinetic-devel branch
- build only rostime, roscpp and ros_tutorials
- run any of the talker/listeners from the tutorials

I built the parts to be tested and used the rest from /opt/ros/kinetic
```
catkin build --no-deps rostime
catkin build --no-deps roscpp
catkin build --no-deps roscpp_tutorials
```
It still works as expected, regardless if I run the listener from the binaries or the rebuilt tutorials...

NickZ, 2018-03-12:
Ok, this is what I did to reproduce consistently:

1. Get a fresh Ubuntu 16.04 image (in either a VM, bare-metal, or a lxc/docker container).
2. Install the ros-base package (`sudo apt-get install ros-kinetic-ros-base`)
3. create a clean catkin workspace based on /opt/ros/kinetic
4. Populate it with `ros_comm`, `roscpp_core`, and `ros_tutorials` repos on the kinetic-devel branch
5. Either run `catkin_make`, or use the build process you describe above, I've reproduced it with both.
6. source devel/setup.bash, run `roscore`, observe the cpu in top.

flixr, 2018-03-12:
Could it be that this is a problem with `catkin_make`, as I was using `catkin` from the `catkin_tools` package...
Does this problem also occur with `catkin_make_isolated`?

NickZ, 2018-03-12:
as I've said, I've done this using both the build process described in your post, just using`catkin`, and with `catkin_make`. I get the same bug. 

It does not occur when I have built the whole ros_comm stack from source, using `catkin_make_isolated`.

flixr, 2018-03-12:
Hm.... I used the install space when testing this... will try again with the devel space.
But this still sounds like basically a known issue with `caktin_make` to me...

NickZ, 2018-03-13:
Just for reference, what catkin bugs should I be aware of?

flixr, 2018-03-13:
Should have phrased it more precisely: known limitation.
Currently can't this anywhere in the catkin docs, but it's nicely summarized here: http://catkin-tools.readthedocs.io/en/latest/migration.html

flixr, 2018-03-14:
Ok, I can reproduce it with the Dockerfile https://gist.github.com/flixr/77e36cfac037700080efd937148b0f7f

To test this
```
docker build -t test_ros_comm .
docker run -it test_ros_comm
```
and observe the high cpu load as you said...

I have no clue why this is happening though.

flixr, 2018-03-14:
It's especially weird because we have over 100 devices running with https://github.com/ros/ros_comm/pull/1250 without problems.
Although that is still on indigo, I just built a new ros-indigo-roscpp debian package on our CI (based on the ros:indigo-perception image) and installed that updated deb in our systems.

fujitatomoya, 2018-10-15:
core tells us the following link problem, we are not sure the root cause of this yet.

when problem observed, libboost_thread is used as following,
which means no pthread_condattr_setclock is issued before.

#0 pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225 
#1 0x00007fcbd757ebdd in boost::condition_variable::do_wait_until (this=0x10022e8, m=..., timeout=...) at /usr/include/boost/thread/pthread/condition_variable.hpp:108 

when problem is NOT observed,
internal class is used with pthread_condattr_setclock.

#0 pthread_cond_timedwait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_timedwait.S:225 
#1 0x00007fc1f5da5501 in boost::condition_variable::do_wait_until (this=0x74c2e8, m=..., timeout=...)
at  /home/sharedata/Projects/ROS1/src/ros_comm/clients/roscpp/include/boost_161_pthread_condition_variable.h:111

fujitatomoya, 2018-10-15:
so far, cannot find root cause.
does anyone have clue for this?

NickZ, 2018-11-03:
@meyerj excellent, thank you for this

fujitatomoya, 2018-11-03:
@meyerj i do agree with the #3 assumption.

meyerj, 2018-12-04:
> @meyerj did you finally fix this issue ?

Yes, but my patch broke ROS timers and I did not have time yet to look into it again. But I could open a PR with the current patch, if someone else is interested.

meyerj, 2018-12-04:
Non-working patch pushed to https://github.com/ros/ros_comm/compare/kinetic-devel...meyerj:fix-1343.

meyerj, 2018-12-06:
Proposed patch submitted as #1557.

closed 2020-02-08 by dirk-thomas