ros/ros_comm #1346: rosout: logging + topic subscription: not thread safe

krixkrix, 2018-03-09:
If a node subscribes to a topic in one thread and logs to rosout in another, a conflict can occur, since the nodes topic list may be altered while the rosout logging traverses it.

The following error is reported:

> Unable to report rosout: Set changed size during iteration

As well as tcpros protocol synchronization is reported lost:

> [ERROR] [1520594758.451498758]
> [/rosout]: a message of over a
> gigabyte was predicted in tcpros. that
> seems highly unlikely, so I'll assume
> protocol synchronization is lost.

> [ERROR] [1520594758.451553409]
> [/rosout]: Exception thrown when
> deserializing message of length [0]
> from [/rosout_test]: Buffer Overrun

> [ERROR] [1520594758.451622886]
> [/rosout]: Exception thrown when
> deserializing message of length [39]
> from [/rosout_test]: Buffer Overrun

This python ROS node reproduces the problem:

    #!/usr/bin/env python
    
    import rospy
    from std_msgs.msg import Float64
    from threading import Thread
    
    def topic_callback(msg):
        pass
    
    def my_topic_thread():
        """endless subscribing and unsubscribing to a list of topics"""
    
        topic_names = ["/some/topic{}".format(i) for i in range(0, 100)]
    
        while not rospy.is_shutdown():
            rospy.loginfo("subscribing")
            subs = []
            for topic in topic_names:
                subs.append(rospy.Subscriber(topic, Float64, topic_callback))
                rospy.sleep(0.001)
    
            rospy.loginfo("unsubscribing")
            for sub in subs:
                sub.unregister()
    
    if __name__ == '__main__':
    
        rospy.init_node('demo')
    
        topic_thread = Thread(target=my_topic_thread)
        topic_thread.start()
    
        while not rospy.is_shutdown():
            # This logging fails intermittently, because the nodes
            # topic list is changed during logging
            # Error: protocol synchronization is lost
            rospy.loginfo("dummy message")
            rospy.sleep(0.001)
    
        topic_thread.join()

kinetic, ubuntu 16.04

gavanderhoorn, 2018-03-09:
From ROS Answers Q&A: [#q284911](https://answers.ros.org/question/284911).

krixkrix, 2018-03-09:
The problem can be solved by copying the topics list before creating the Log object, though I don't know if this is the proper way.
Patch:

```
--- rosout.py.org	2018-03-09 13:56:23.761524171 +0100
+++ rosout.py	2018-03-09 13:56:53.989569745 +0100
@@ -85,7 +85,7 @@
                 try:
                     _in_rosout = True
                     msg = str(msg)
-                    topics = get_topic_manager().get_topics()
+                    topics = get_topic_manager().get_topics().copy()
                     l = Log(level=level, name=str(rospy.names.get_caller_id()), msg=str(msg), topics=topics, file=fname, line=line, function=func)
                     l.header.stamp = Time.now()
                     _rosout_pub.publish(l)

```

pbaughman, 2018-05-31:
Just got burned by this yesterday

dirk-thomas, 2018-05-31:
@krixkrix Please consider to create a pull request with the proposed patch targeting the default branch (currently `melodic-devel`).

pbaughman, 2018-05-31:
I can confirm that making a copy of the set that's returned by get_topics keeps this from blowing up for us.

pbaughman, 2018-05-31:
@dirk-thomas I put krixkrix's proposed fix here: https://github.com/ros/ros_comm/pull/1416 for melodic.  Is there a separate process to back-port the fix?  It's the same seven characters in Indigo, Kinetic, etc. . .

dirk-thomas, 2018-05-31:
> I put krixkrix's proposed fix here: #1416 for melodic.

Thanks.

> Is there a separate process to back-port the fix?

Before a new patch release is being made into "older" distros all patches from the newer distro are being considered. Feature is commonly not backported but bug fixes with a low potential for regression (like this one) are usually backported.