ros/ros_comm #1347: rostopic: pub a single message from file

m-naumann, 2018-03-12:
Hello,

when trying to publishing a single message from file, it is published only once, even though a rate is given.

It is certainly desired that, when having multiple messages in the file, they are published one after another at the desired rate. But when providing just one message, it should be repeatedly published.

### Example:

#### as desired, `argv_publish`:

`rostopic pub -r 1 /test std_msgs/Header '{}'`
-> publishes a header every second

#### misleading, from file via `stdin_publish`
`rostopic pub -r 1 /test std_msgs/Header -f header.yaml`

with header.yaml:

```
  seq: 0
  stamp:
    secs: 0
    nsecs:         0
  frame_id: 'test_frame'
```
-> publishes the message once in latching mode

koenlek, 2019-02-08:
Same issue here

dirk-thomas, 2019-03-09:
Addressed in #1635.

closed 2019-03-09 by dirk-thomas