ros/ros_comm #1351: rospy can only support one subscriber type per topic

tfoote, 2018-03-16:
If I create two subscribers for a single topic with different types. The 2nd callback gets the type of the first registered callback. 

This is a refiling of: https://github.com/ros/geometry2/issues/288

Reproduction script
```
#!/usr/bin/env python
import rospy
from tf2_msgs.msg import TFMessage
import tf2_ros

def callback(data):
    rospy.loginfo(rospy.get_caller_id() + "I heard something...AnyThing?")
    print("Type of data is", type(data))

def typed_callback(data):
    who = data._connection_header.get('callerid', "default_authority")
    print("Type of data is", type(data))
    for transform in data.transforms:
        print("Iterating data transforms", transform)

def topic_subscriber():
    sub2 = rospy.Subscriber("/tf", TFMessage, typed_callback)
    sub1 = rospy.Subscriber("/tf", rospy.AnyMsg, callback)

if __name__ == '__main__':
    rospy.init_node('test_callback_types', anonymous=True)
    topic_subscriber()
    rospy.spin()
```

To see the errors run the above with this data source

```
$ rostopic pub /tf tf2_msgs/TFMessage "transforms:
- header:
    seq: 0
    stamp:
      secs: 0
      nsecs: 0
    frame_id: ''
  child_frame_id: ''
  transform:
    translation:
      x: 0.0
      y: 0.0
      z: 0.0
    rotation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 0.0"  -r 10
```

Depending on the order of creating the Subscribers you will get both callback in either `AnyMsg` or `TFMessage` regardless of what the prototype you subscribed with.

```
[INFO] [1521221823.348291]: /test_callback_types_7127_1521221822176I heard something...AnyThing?
('Type of data is', <class 'tf2_msgs.msg._TFMessage.TFMessage'>)
('Type of data is', <class 'tf2_msgs.msg._TFMessage.TFMessage'>)
('Iterating data transforms', header: 
  seq: 0
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
child_frame_id: ''
transform: 
  translation: 
    x: 0.0
    y: 0.0
    z: 0.0
  rotation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 0.0)
[INFO] [1521221823.448275]: /test_callback_types_7127_1521221822176I heard something...AnyThing?
('Type of data is', <class 'tf2_msgs.msg._TFMessage.TFMessage'>)
```
vs
```
[INFO] [1521221815.948104]: /test_callback_types_6931_1521221811780I heard something...AnyThing?
('Type of data is', <class 'rospy.msg.AnyMsg'>)
('Type of data is', <class 'rospy.msg.AnyMsg'>)
[ERROR] [1521221815.948586]: bad callback: <function typed_callback at 0x7f072d586c08>
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "test.py", line 14, in typed_callback
    for transform in data.transforms:
AttributeError: 'AnyMsg' object has no attribute 'transforms'
```

dirk-thomas, 2018-03-16:
All entities created for the same topic do share the same topic instance under the hood. Therefore the types can currently not be different. I don't see any easy way to address this without refactoring the underlying implementation.

Please consider to provide a pull request for this since it is unlikely that the maintainers will have time to look into this.

lucasw, 2018-03-23:
A workaround is to make both subscribers AnyMsg but deserialize in the callback where it is needed:

        data = TFMessage().deserialize(anydata._buff)

But that doesn't work for ros/geometry2#288 or other cases where one of the subscribers is buried in a library.

MichaelGrupp, 2020-05-07:
I stumbled upon this today and suggest to add at least a warning: https://github.com/ros/ros_comm/pull/1947

closed 2025-05-31 by sloretz