ros/ros_comm #1353: Add a reindex method to rosbag::Bag

bobbrady, 2018-03-21:
I am writing commercial software to support extracting ROS bag info for debugging robot issues.  C++ functionality is desirable for runtime speed and also to simply distribute binary executables plus shared objects for development and prod deploy.

I was able to implement a rosbag-to-json converter in C++, but I am not able to use to the ROS C++ API code as-is to reindex with C++.  The reason is Bag::readFileHeaderRecord method throws a BagUnindexedException at [line 394](https://github.com/ros/ros_comm/blob/1fa062ca269e5870b0c48c8b1d484890ae393b1b/tools/rosbag_storage/src/bag.cpp#L393line). This prevents the use of [Bag::startReadingVersion102](https://github.com/ros/ros_comm/blob/c99fa76b1edc3417aa1404f40f4ef452ee662f9f/tools/rosbag_storage/src/bag.cpp#L305) or [Bag::startReadingVersion200](https://github.com/ros/ros_comm/blob/c99fa76b1edc3417aa1404f40f4ef452ee662f9f/tools/rosbag_storage/src/bag.cpp#L270) methods to generate all bag index information by rereading the message records, like the corresponding python [_BagReader102_Unindexed](https://github.com/ros/ros_comm/blob/b431e228f86e241fc157ad21ad0c47b5ee39edd0/tools/rosbag/src/rosbag/bag.py#L1734) and [_BagReader200_Unindexed](https://github.com/ros/ros_comm/blob/b431e228f86e241fc157ad21ad0c47b5ee39edd0/tools/rosbag/src/rosbag/bag.py#L2074) can do.  The python code also provides several helper methods to support the reindex operation.

I'm requesting the addition of a Bag::reindex function in the C++ API, [similar to what is provided for in the python Bag class](https://github.com/ros/ros_comm/blob/b431e228f86e241fc157ad21ad0c47b5ee39edd0/tools/rosbag/src/rosbag/bag.py#L402).  Right now, I'm using the C++ bag-to-json converter binary for speed and the python rosbag utility for reindex functionality.  It would be better to just use all C++.

mikepurvis, 2018-03-21:
The maintainers are unlikely to work on this ourselves in the near term, but we'd be delighted to review a contribution which added the suggested functionality.

harlowja, 2019-07-14:
Enjoy, https://gist.github.com/harlowja/405e255065f3236b4d0db5a8d94d1825

harlowja, 2019-07-14:
(I don't have enough time to make a PR, but someone else can).

closed 2025-05-31 by sloretz