ros/ros_comm #1362: Message Size-dependent Queueing/Latency

shomin, 2018-04-09:
Originally posted this on ROS answers, but this appears to be a ros_comm bug
https://answers.ros.org/question/287292/message-size-dependent-queueinglatency/

I've noticed some delays up to 1/(publish_frequency) for medium sized messages.  I've reproduced the behavior just using rostopic below:

    rostopic pub -r 50 /test std_msgs/String "data: '`printf "%0.s-" {1..100}`'"
    rostopic hz -w 50 /test

    average rate: 49.983
    	min: 0.018s max: 0.022s std dev: 0.00043s window: 50

Note the printf command just makes the data 100 hyphens.  Pretty normal, what you might expect. Now how about 1000 bytes of hyphens:

    rostopic pub -r 50 /test std_msgs/String "data: '`printf "%0.s-" {1..1000}`'"

    average rate: 50.033
    	min: 0.019s max: 0.020s std dev: 0.00018s window: 50
    average rate: 49.994
    	min: 0.019s max: 0.021s std dev: 0.00019s window: 50
    average rate: 49.990
    	min: 0.019s max: 0.021s std dev: 0.00021s window: 50
    average rate: 49.973
    	min: 0.019s max: 0.021s std dev: 0.00021s window: 50
    average rate: 49.027
    	min: 0.000s max: 0.040s std dev: 0.01368s window: 50
    average rate: 51.041
    	min: 0.000s max: 0.041s std dev: 0.01996s window: 50
    average rate: 51.035
    	min: 0.000s max: 0.044s std dev: 0.01963s window: 50

Notice that after about 3 seconds, we see a min of 0 seconds and a max of .04 seconds. Introspecting, after this starts to happen, it continues happens very consistency (every other message is queued). 

Now for the extra weirdness, this doesn't seem to happen with big messages.  100,000 bytes of hyphens:

    rostopic pub -r 50 /test std_msgs/String "data: '`printf "%0.s-" {1..100000}`'"
    
    average rate: 50.037
    	min: 0.019s max: 0.020s std dev: 0.00020s window: 50
    average rate: 49.994
    	min: 0.019s max: 0.021s std dev: 0.00027s window: 50
    average rate: 49.979
    	min: 0.019s max: 0.021s std dev: 0.00020s window: 50
    average rate: 50.020
    	min: 0.018s max: 0.022s std dev: 0.00042s window: 50
    average rate: 49.995
    	min: 0.019s max: 0.021s std dev: 0.00020s window: 50
    average rate: 50.007
    	min: 0.019s max: 0.021s std dev: 0.00017s window: 50

So, no problem for large messages!  At 50 Hz, the size of message that seems to do this is EXACTLY 516 bytes to 65475 bytes of data.  Try this out:

    rostopic pub -r 50 /test std_msgs/String "data: '`printf "%0.s-" {1..515}`'"

^ No problem

    rostopic pub -r 50 /test std_msgs/String "data: '`printf "%0.s-" {1..516}`'"

^ Problem!

    rostopic pub -r 50 /test std_msgs/String "data: '`printf "%0.s-" {1..65475}`'"

^ No problem

    rostopic pub -r 50 /test std_msgs/String "data: '`printf "%0.s-" {1..65574}`'"

^ Problem!

Unfortunately for my use case, I can't just make all my messages bigger (or smaller).  Introspecting in the specific use case, it seems like queueing in that there will be a miss and then 2 messages the next expected cycle.  However, after it starts, this continues to happen with regularity, unless the messages are bigger or smaller.  Any idea why this is happening?

I've reproduced in roscpp and rospy, with and without tcdnodelay, zero queues, and sychonous publishing (queu_size=None).  It always exhibits this same behavior.  I've also tried this in the indigo-ros-core docker container as well as the lunar-ros-core docker container.  Lastly, I've also adjusted process priority, which also has no effect. Obviously, this is all on one machine as well, no network issues here.

gavanderhoorn, 2018-04-10:
@shomin wrote:
> this appears to be a ros_comm bug

it might be a little early to classify this as a bug (I also don't believe we did that in the ROS Answers post you link to). It could be that this is known behaviour, but un(der)documented or an unfortunate result of certain design decisions.

In any case it seems like something worthwhile to figure out. And since you mention that you've reproduced this in both `rospy` as well as in `roscpp`, I thought it might be ok to post this as an issue here.

scott-eddy, 2018-04-10:
Note that is is an issue with [Nagle's algorithm](https://en.wikipedia.org/wiki/Nagle%27s_algorithm) on the Linux TCP stack which is used by default between ROS publisher's and subscribers.  If you create a minimal Publisher/Subscriber node specifying [ros::TransportHints::tcpNoDelay()](http://docs.ros.org/kinetic/api/roscpp/html/classros_1_1TransportHints.html#a03191a9987162fca0ae2c81fa79fcde9) you will see that the queueing stops.  

I've never tried turning off the algorithm globally on a system [but it would appear possible](https://stackoverflow.com/questions/17842406/how-would-one-disable-nagles-algorithm-in-linux)

gavanderhoorn, 2018-04-10:
I'm now responsible for this cross-post and immediately bitten by it: @shomin mentioned that he's already disabled Nagle:

> I've reproduced in roscpp and rospy, with and without tcdnodelay,

could still be Nagle, but that's something the OP would need to verify then.

gavanderhoorn, 2018-04-10:
`libros` can be compiled to disable Nagle always (so irrespective of transport hints):

https://github.com/ros/ros_comm/blob/6bb54460f31527718d539fa7ef076d5cf51ffcd8/clients/roscpp/src/libros/transport/transport_tcp.cpp#L132-L134

@shomin: perhaps you could see whether this changes the behaviour you've observed.

mikeshomin, 2018-04-10:
I was about to post the python source of the test, which uses tcp_nodelay, but you actually made me realize I hadn't done it correctly.  I was under the impression that if the publisher was using tcp_nodelay, that the subscriber would use it too, but it appears that both the publisher and subscriber (or at least the subscriber) need to individually set this.  Using tcp_nodelay on both publisher and subscriber works for me and eliminates this problem.  I will close this unless you think there is still anything to address.

mikeshomin, 2018-04-10:
Thank you both for your help

scott-eddy, 2018-04-10:
Glad it worked.  Can you please accept the answer on ros answers so others know that is the resolution

scott-eddy, 2018-04-10:
@gavanderhoorn worth closing this

gavanderhoorn, 2018-04-11:
I'm not a maintainer here, only @shomin or one of the admins on this repository can close it :)

closed 2018-04-11 by shomin