ros/ros_comm #1366: Demux crashes when using __none

lfdebrux, 2018-04-16:


closed 2018-04-24 by mikepurvis

peci1, 2020-02-18:
Argh... bumped into this now on melodic... When will there finally be a new release of ros_comm???

peci1, 2020-02-18:
For others who suffer by this bug: as a workaround, you can always add a "blackhole" topic which nobody subscribes, and redirect the output to it instead of `__none`.

dirk-thomas, 2020-02-19:
> Argh... bumped into this now on melodic... When will there finally be a new release of ros_comm???

This issue was closed when the referenced commit was merged: 2b3f43d on Apr 24, 2018. And that commit has been released for a very long time into Melodic (around May 2018). So if you still experience issues you might want to double check if there is an open ticket describing it or create a new issue.

peci1, 2020-02-19:
You're right, in the end it looks like a different issue, but with very similar symptoms. See #1889 .

commit 2b3f43d4 (2018-04-24): topic_tools: Check that output topic is valid in demux. Fixes #1366 (#1367)
commit 017ce50c (2018-08-17): topic_tools: Check that output topic is valid in demux. Fixes #1366 (#1367)
commit 0ab0bc3c (2018-08-23): topic_tools: Check that output topic is valid in demux. Fixes #1366 (#1367)