ros/ros_comm #1369: double dash in roslaunch comment

dogoepp, 2018-04-19:
In the following sample, I put a backtick in a comment within a launch file. This triggers the error below.

```xml

<launch>
  <!--
    What is done here
      - load URDF description the TF tree
      - control node for the arm (dynamixel hardware interface)
      - control node for the YouBot mobile base (youbot driver)

    Arguments
      The arguments of this launch file can be described with
      `roslaunch omni_bringup omnigrasper.launch --ros-args`
  -->

  <node whatever.../>
</launch>
```

Error message

> Invalid roslaunch XML syntax: not well-formed (invalid token): line 40, column 51
> The traceback for the exception was written to the log file

With some tests, I isolated the cause to be the double dashes.

dirk-thomas, 2018-04-19:
As the XML specification states in [2.5 Comments](https://www.w3.org/TR/REC-xml/#sec-comments):

> the string " -- " (double-hyphen) must not occur within comments.

You simply can't put a double hyphen in the comment.

closed 2018-04-19 by dirk-thomas

dogoepp, 2018-04-20:
Thanks, I had not imagined that this could be forbidden.

samarth-robo, 2019-07-09:
See [this article](https://norlab.ulaval.ca/research/how-to-comment-lauch-file/), apparently you can use `<?ignore <stuff> ?>`, and that works with ` comments inside it too.