ros/ros_comm #1370: gdb debugging generates error messages "[ERROR] [<TIME_STAMP>]: poll failed with error Interrupted system call"

alrros, 2018-04-20:
summary:
--------
In the current version of ros_comm, when debugging with gdb using either single-steps or breakpoints, the error message "[ERROR] [<TIME_STAMP>]: poll failed with error Interrupted system call" is generated repeatedly.
The error messages were not generated in older ros_comm versions.
Only since end of february / beginning of march 2018 this problem seem to have occured.

severity of problem:
--------------------
A lot of error messages are generated during debugging.
Important error messages are therefore harder to see and may be overseen.

setup:
------
Ubuntu 16.04.4 LTS
ros-kinetic
ros-kinetic-roscpp_1.12.13-0xenial-20180222-205956-0800_amd64
gdb 7.11.1
gdbserver 7.11.1

details to reproduce problem:
-----------------------------
Debugging was performed with two simple publisher/subscriber nodes similar to talker/listener nodes of roscpp_tutorials.
Eclipse was used as main IDE.
See reference [1] how to setup eclipse project from catkin workspace.
(Note: problem can also be reproduced with command line gdb without eclipse).
The behaviour of error message generation dependends on different debugging setups.
The following list provides an overview of debugging setups (gdb with eclipse) and the occurance of error messages.

    Local C/C++ Application:
    - all-stop-mode
        -- error messages on single-step
    - non-stop-mode
        -- no error messages on single-step nor on breakpoints
        -- error messages, when thread #2 is suspended during poll and then resumed

    Remote C/C++ Application on localhost (with gdbserver):
    - non-stop-mode
        -- error messages on single-steps and on breakpoints

    Remote C/C++ Application on remote machine (with gdbserver):
    - non-stop-mode
        -- no error messages on single-steps nor on breakpoints
        -- no error, when thread #2 is suspended during poll and then resumed
    - non-stop-mode (having built ros_comm libroscpp.so in debug-mode)
        --- error messages on single-steps and on breakpoints
        

        
hints:
------
Building ros_comm package in debug mode, the source of the error messages was locatable.
See reference [2] how to build ros from source.
The library libroscpp.so generates the error messages.
The problem is located in the following files, lines and functions (as of writing 2018-04-20 16:00):

    - https://github.com/ros/ros_comm/blob/kinetic-devel/clients/roscpp/src/libros/io.cpp 
        -- line: 
            --- ~165
            --- 304 - 316
        -- function:
            --- pollfd_vector_ptr poll_sockets(int epfd, socket_pollfd *fds, nfds_t nfds, int timeout)
    - https://github.com/ros/ros_comm/blob/kinetic-devel/clients/roscpp/src/libros/poll_set.cpp
        -- line:
            --- ~184
        -- function:
            --- void PollSet::update(int poll_timeout)
            
In the simple publisher/subscriber test applications, the above mentioned functions were executed in thread #2.
During debugging the main thread, in io.cpp, function poll_sockets, the system call epoll_wait is interrupted.

    - io.cpp line 306:    int fd_cnt = ::epoll_wait(epfd, ev, nfds, timeout);

A comment indicates, that an interrupted system call is not an error.

    - io.cpp line 310:    // EINTR means that we got interrupted by a signal, and is not an error

The gdb online documentation also states, that interrupted system calls are not errors, see reference [3].
The ROS_ERROR macro is NOT called in io.cpp.
But in poll_set.cpp, function PollSet::update (the caller of io.cpp, function poll_sockets), the return value of poll_sockets in case if interrupted system call is still handled as an error and the macro ROS_ERROR_STREAM is called.
This leads to many error messages with "poll failed with error Interrupted system call" during debugging.

The issue may be fixable by wrapping the system call epoll_wait into a while loop, see reference [4][5].

references:
-----------
references for setup:
    [1] http://wiki.ros.org/IDEs#Eclipse
    [2] http://wiki.ros.org/Installation/Source

further references about interrupted system calls can be found in gdbs online documentation:
    [3] https://sourceware.org/gdb/current/onlinedocs/gdb/Interrupted-System-Calls.html#Interrupted-System-Calls

similar issue of interrupted system calls and approach to solve
    [4] https://stackoverflow.com/questions/6813629/how-do-i-use-gdb-for-multi-threaded-networking-program
    [5] https://stackoverflow.com/a/6815333

GitBubble, 2018-06-14:
great point

rojas70, 2018-10-16:
will this issue get pulled soon?

llvs, 2018-10-17:
This error also appears when debugging with GDB integrated in QtCreator.

tahsinkose, 2018-12-11:
Same from VSCode integrated GDB debugger. `advertiseService` function call is directly bypassed due to this error.

bbferka, 2018-12-30:
does anyone know of a temporary solution for this? I am using qtcreator to debug my nodes and these messages are really annoying.

vishnureghu007, 2019-03-20:
Any update reagarding this issue??

julled, 2019-04-06:
I am also waiting for a bugfix. Got that problem when using gdb in qtcreator.

cognitiveRobot, 2019-04-17:
I am getting the same error. Any update?
```
OS: Ubuntu 18.04 (4.15.0-47-generic)
IDE: VSCode 1.33.1
Debugger: GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
```

PR #1868 (2020-02-10): Do not display error message if poll yields EINTR
Closes #1370
closes TonyRobotics/RoboWare#63
closes ericsantii/alexa-turtlesim-ros#2
commit 2e994097 (2020-02-10): Do not display error message if poll yields EINTR (#1868)

* Do not display error message if poll yields EINTR

Closes #1370
closes TonyRobotics/RoboWare#63
closes ericsantii/alexa-turtlesim-ros#2

* Keep abstraction compatibility
commit 5b6cf04b (2020-08-04): Do not display error message if poll yields EINTR (#1868)

* Do not display error message if poll yields EINTR

Closes #1370
closes TonyRobotics/RoboWare#63
closes ericsantii/alexa-turtlesim-ros#2

* Keep abstraction compatibility
closed 2020-02-10 by dirk-thomas