ros/ros_comm #1377: Alternatives to rospy publish and subscribe

chrisspen, 2018-04-25:
This is more of a general question and/or feature request, so I apologize if this isn't the right place. However, as it involves a possible pull-request, I felt it better to submit here rather than answers.ros.org.

I've gone through all the Rospy tutorials and have written a lot of my own Rospy-based nodes, and I'm hoping you can clarify something.

One of the slightly frustrating things I've found about Rospy, and much of the topic-sending command line methods, is the requirement that you explicitly state the type of the data you're sending. Is there a need for this? As far as I can tell, ROS doesn't support topic "overloading", and it already knows the message types that each topic expects. Why then do all the topic based methods require the user to explicitly cast their data to the expected type?

This became so frustrating to me that I wrote simple wrappers around the standard `rospy.Publisher()` and `rospy.Subscriber()` methods that use introspection to lookup the type and do some other boilerplate automatically. So, for example, instead of doing:

    import rospy
    import mynode.msgs import MyMsg
    pub = rospy.Publisher('/mynode/mytopic', MyMsg, queue_size=1)
    msg = MyMsg()
    msg.field1 = 123
    msg.field2 = 456
    msg.field3 = 'abc'
    pub.publish(msg)

you can just do:

    from ros_wrappers import publish
     publish('/mynode/mytopic', 123, 456, 'abc')

I've found this drastically simplifies my code and makes it much easier to read. However, I can't help but think I'm missing something, otherwise it would have already been designed this way. What am I missing?

Is there any interest in a pull-request for these wrappers, or would this be outside the scope of ros_comm?

dirk-thomas, 2018-04-27:
In your example time is not being considered. Between the creation of a publisher and the actual `publish` call an arbitrary amount of time could pass. During that time the publisher is already known in the ROS graph and its type can be queried even though no message has been sent yet.

In you short version (which only shows the function call but not how the function is implemented) the ownership and lifetime of the publisher instance are unclear. If they are not stored anywhere the Python garbage collection might delete the publisher anytime.

You example only publishing a single message is not very realistic. Commonly your single message wouldn't be received by any subscriber because it takes a non-zero amount of time for existing subscribers to connect to a newly created publisher. So your initial message might just get "lost"
 because it is published before the subscribers had a chance to establish a connection.

chrisspen, 2018-04-27:
>Commonly your single message wouldn't be received by any subscriber because it takes a non-zero amount of time for existing subscribers to connect to a newly created publisher. So your initial message might just get "lost"

I had this same concern, which is why I keep a cache of Publisher instances in a global hash table. However, even before I implemented that, I never noticed any missing messages, so that's seems more of a theoretical concern. Python's garbage collection is quite lazy, and ROS typically sends the message right away.

dirk-thomas, 2018-04-27:
> I never noticed any missing messages, so that's seems more of a theoretical concern.

It is actually a fairly frequent question, e.g. the latest one: https://answers.ros.org/question/287548/ros_tutorials-roscpp-talkerlistener-loses-first-message-or-two/

> ROS typically sends the message right away.

The problem is not the time it takes to send the message. The subscriber just didn't have enough time to establish a connection yet.

Anyway getting back to your original question: I don't think it makes sense to add wrapper functions as described into `ros_comm` (due to the mentioned concerns).

chrisspen, 2018-04-27:
Which concern are you referring to? You're not convinced my implementation won't be garbage collected?

chrisspen, 2018-04-27:
I guess I still don't see how your concerns uniquely apply to my implementation, as it's functionally identical to the status quo, as I illustrated earlier. If messages are currently being dropped, then why is it a concern that this still happens with my wrapper?

And the global storage can easily be accessed like any other global method or function in rospy. For example, `rospy.init_node(name)` doesn't return a reference to a Node instance. It auto-magically configures various global objects to register the current script in the ROS network. My wrappers only simply the interface by making it less verbose. They don't change the underlying functionality.

tfoote, 2018-04-28:
> And the global storage can easily be accessed like any other global method or function in rospy.

One use of globals does not make it good practice to use globals everywhere else. And we don't design into our public API something that requires the users to reach into the internals of those global values to be able to do an expected functionality. If we were to write this again. (see rclpy) we wouldn't use the globals.

This is also a symmetry point to our other APIs, and keeps the developer more aware of the resource allocations and lifetimes that they are incurring. And provides visibility into the timing between the first publish call and the publisher creation to control it to not lose the first message it's much clearer.

Your presentation shows it being 5 lines to create the message, but you can still use the implicit constructor arguments. Which just means that you have 2 lines, one to create the publisher and one to use it. However for good practice the first should be in a different scope, often in a class constructor and the second inside the loop or callback where it will be used, thus although there is one more line the overhead on the developer is not that significant.

dirk-thomas, 2018-04-28:
Actually for the propose code:

> publish('/mynode/mytopic', 123, 456, 'abc')

The type of the topic is nowhere defined. So that wouldn't work.

And for the initialization of the message we intentionally don't use positional arguments since that code would break if the message definition is updated in the future.

chrisspen, 2018-04-28:
>The type of the topic is nowhere defined. So that wouldn't work.

That's by design and was my earlier point. ROS already knows what the type is, so explicitly specifying the type is redundant and serves no purpose. I'm using this code, so I'm not sure why you would say it "wouldn't work". I assure you, it works. I have it running on a physical robot.

dirk-thomas, 2018-04-28:
Please share a working code snippet as well as steps to reproduce.

dirk-thomas, 2018-04-28:
> ROS already knows what the type is, so explicitly specifying the type is redundant and serves no purpose.

If there is no other publisher / subscriber for that topic where would the information come from?

Please share a working code snippet as well as steps to reproduce if you are convinced otherwise.

chrisspen, 2018-04-30:
>If there is no other publisher / subscriber for that topic where would the information come from?

I think I see the misunderstanding. Yes, there needs to be at least one publisher or subscriber that explicitly defines its type. However, all others can then use introspection to lookup this type. Requiring every publisher and subscriber to define the message type violates the DRY principle, adds extra work for the developer, and leads to more maintenance down the road. I'd love to see topic name strings be replaced with global static variables, in a manner similar to how message types are defined, but that's a different issue.

Here's my code:

    import importlib
    from commands import getoutput

    import rospy

    _publisher_cache = {} # {topic: publisher_instance}

    def get_published_topics(target=None):
        """
        Retrieves a hash of topics of the form {topic: type}.
        """
        topic_types = {}
        if target:
            raw_topics = getoutput('rostopic list %s --verbose' % target).split('\n')
        else:
            raw_topics = getoutput('rostopic list --verbose').split('\n')
        for raw_topic in raw_topics:
            parts = raw_topic.split(' ')
            if len(parts) == 6 and parts[1] == '*':
                _, _, topic, type_name, cnt, io_type = parts
                type_name = type_name[1:-1]
                topic_types[topic] = type_name
        return topic_types

    def subscribe(topic, cb):
        """
        Dynamically looks up the subscriber for a topic, as well as the type class, and subscribes.
        """
        topic_types = get_published_topics(topic)
        if topic not in topic_types:
            all_topic_types = get_published_topics()
            raise Exception('Topic "%s" is not one of the valid published topics: %s' % (topic, ', '.join(sorted(all_topic_types))))
        typ = topic_types[topic]
        typ_module_name, typ_cls_name = typ.split('/')
        typ_cls = getattr(importlib.import_module('%s.msg' % typ_module_name), typ_cls_name)
        rospy.loginfo('subscribing to: %s %s %s', topic, typ_cls, cb)
        return rospy.Subscriber(topic, typ_cls, cb)

    def publish(topic, data=None, queue_size=1):
        """
        Dynamically looks up the publisher for a topic, as well as the type class, wraps the data in the class, and then publishes it.
        """
        topic_types = get_published_topics(topic)
        if topic not in topic_types:
            all_topic_types = get_published_topics()
            raise Exception('Topic "%s" is not one of the valid published topics: %s' % (topic, ', '.join(sorted(all_topic_types))))
        typ = topic_types[topic]
        typ_module_name, typ_cls_name = typ.split('/')
        typ_cls = getattr(importlib.import_module('%s.msg' % typ_module_name), typ_cls_name)
        if topic not in _publisher_cache:
            _publisher_cache[topic] = rospy.Publisher(topic, typ_cls, queue_size=queue_size)
        pub = _publisher_cache[topic]
        if typ == 'std_msgs/Empty':
            data = typ_cls()
        else:
            data = typ_cls(data)
        rospy.loginfo('publishing %s to %s', data, topic)
        pub.publish(data)

Also, I think I stumbled across a bug in `rostopic list --verbose` while developing  this. Any relative topics defined from a subscriber in rosserial that contain a "/" report their type as "Empty". When I replaced the "/" with "_", then rostopic reported the correct message type.

dirk-thomas, 2018-04-30:
> I think I see the misunderstanding. Yes, there needs to be at least one publisher or subscriber that explicitly defines its type. However, all others can then use introspection to lookup this type.

The point is that ROS doesn't rely on a specific order of starting your nodes. E.g. in a launch file you can't even specify any order. Therefore I don't think we should encourage that kind of use by adding this "shortcut" API.

> I stumbled across a bug in rostopic list --verbose while developing this

Please fill a separate ticket about it and consider creating a pull request to address it if possible.

dirk-thomas, 2018-05-01:
From my point of view the proposed API has concerns and doesn't provide much value (condensing two lines for creating the publisher and publishing a message into a single function while hiding the publisher instance in global variable). Therefore I don't a strong reason for it being integrated into `rospy`.

That being said I encourage you to package the proposed API into a ROS package to make it available to others. If it turns out to be as valuable as you expect it to be users will just pick it up and use it.

chrisspen, 2018-05-04:
I understand. Thanks for talking through this with me.

closed 2018-05-04 by dirk-thomas