ros/ros_comm #1383: Problem compiling `utilities/message_filters/synchronizer.h` with GCC 8.0.1 on Fedora 28

nordmoen, 2018-05-02:
I first reported this issue [here](https://github.com/ros-perception/image_common/issues/81), but the solution seems to be to change [`synchronizer.h:358`](https://github.com/ros/ros_comm/blob/melodic-devel/utilities/message_filters/include/message_filters/synchronizer.h#L358).

The problem that appeared with Fedora 28 and GCC `8.0.1` is:
```
Repos/ros_catkin_ws/src/image_common/image_transport/src/camera_subscriber.cpp:112:64:   required from here
Repos/ros_catkin_ws/install_isolated/include/message_filters/synchronizer.h:358:14: error: invalid operands of types ‘<unresolved overloaded function type>’ and ‘int’ to binary ‘operator<’
     this->add<i>(evt);
```

According to [this StackOverflow answer](https://stackoverflow.com/a/44640490), `synchronizer.h` line 358 is ill-formed. The solution is to add `template` so the line reads `this->template add<i>(evt);`. This is the same syntax used on [line `342`](https://github.com/ros/ros_comm/blob/melodic-devel/utilities/message_filters/include/message_filters/synchronizer.h#L342) so it appears a similar issue has been encountered before.

After applying the modest change, I was able to compile all the packages I needed on Fedora 28 :smiley:

mikepurvis, 2018-05-02:
Please send a PR for this improvement, thanks!

mikepurvis, 2018-05-04:
Perfect, thanks!

closed 2018-05-04 by mikepurvis

lucbettaieb, 2019-04-26:
https://github.com/ros/ros_comm/pull/1705