ros/ros_comm #1384: rospy disables standard python logging

oroulet, 2018-05-03:
try following:

    logging.basicConfig(level=logging.WARN)
    logger.warning(1)
    print(logger)
    rospy.init_node("whaterver")
    logger.warning(2)

the second warning does not get printed whil it sould. I understand that rospy is messing with the normal logging but when using thirs party libraries that relies on logging to print importatnt message to interface this is a big issue. How can I reenable logging for this libraries?

Generally I suppose rospy should not change logging from other libraries unless explicitly requested,

oroulet, 2018-05-03:
I found out I could re-enable logging by adding a root handler:

def fix_logging(level=logging.WARNING):
    console = logging.StreamHandler()
    console.setLevel(level)
    logging.getLogger('').addHandler(console)
    formatter = logging.Formatter('%(levelname)-8s:%(name)-12s: %(message)s')
    console.setFormatter(formatter)
    logging.getLogger('').addHandler(console)

but then rospy logging is printed twice

warpdriv, 2018-05-29:
Having this problem too. Found this work around.

https://gist.github.com/nzjrs/8712011

asimonov, 2018-08-29:
i have hit this problem too and it took me a while to realise ROS is taking over python logging.
I would say it is a bug, not a feature.
What is the reason to do it this way?

sjhansen3, 2018-10-23:
any new info on this?

sjhansen3, 2018-10-23:
I was able to get the logger used by ros with the following line:
logging.getLogger('rosout')

Publishing to this logger works just fine, although other loggers are still screwed up because they are not using this namespace. I agree this is an undesirable situation. It would be nice if existing logging would work along side ros code.

sjhansen3, 2018-10-23:
I did some more testing and digging and testing. 

It is not possible to set the logging level for 'rosout' with the following command:

` logging.getLogger('rosout').setLevel(logging.DEBUG)`

ROS seems to ignore this.

I am however able to add filters on the logging events

mgrrx, 2018-10-23:
As outlined here http://wiki.ros.org/rospy/Overview/Logging#Advanced:_Override_Logging_Configuration you can set the environment variable ROS_PYTHON_LOG_CONFIG_FILE and point it to a file which configures python logging for your needs. You can find the default config in $ROS_ETC_DIR/python_logging.conf
rospy.init_node loads this file for you.

oroulet, 2018-10-23:
@mgrrx this is irrelevant for that big isn't it? The point here is that ros breaks everything else logging by default. This is clearly bad practice

sjhansen3, 2018-10-23:
I guess, I might be missing something but I would tend to agree with @oroulet. Could you explain a bit more how we can configure the ROS_PYTHON_LOG_CONFIG_FILE so that other logging handlers will work alongside ros's use of the python logger? Perhaps there is some key architectural reason why this choice was made and we just need to understand it better- @mgrrx :)

ablakey, 2019-02-20:
I took the workaround shared by @warpdriv and wrapped it to make its usage a bit more obvious: https://gist.github.com/ablakey/4f57dca4ea75ed29c49ff00edf622b38

InspireX96, 2019-08-23:
Found this issue when writing a rostopic monitoring program. After a whole morning of debugging, we found the easiest solution would be reload the logging module after ros_node init:
```
rospy.init_node('blahblah', anonymous=True)
reload(logging)
```
Then we can re-handle logging to python

hd71205991, 2019-11-27:
Actually rospy just read in a logging config file located in $ROS_ROOT/../../etc/ros/python_logging.conf. It sets the root logger's behavior. I made the normal logging function work without interfering rospylog by adding this:
`  customlogger = logging.getLogger('__name__')
    fh = logging.FileHandler("/home/username/mylog.log")
    customlogger.setLevel(logging.DEBUG)
    customlogger.addHandler(fh)
    customlogger.debug("This is for debug")
` 
Remember to call it **AFTER**  the rospy.init_node is called and you are all set. You can add handlers and formatter etc to the logger object and can have the full logging function back.

dadwin, 2020-05-25:
another possible approach is to create your logger under `rosout` logger:
`logger = logging.getLogger(f'rosout.{__name__}')`

then all subsequent logger's calls will print messages into console, log file and `rosout` topic.

130s, 2022-04-05:
I guess there are 2 different usecases are discussed in this thread so far.
- UC-1. Python library `logging` not logging/printing (I guess this is what OP reported about).
- UC-2. `rospy` not including what Python library `logging` logs (? I may be wrong. But this seems to be what multiple people is complaining about).

I'm blocked by UC-1 not realizing. I tried https://github.com/ros/ros_comm/issues/1384#issuecomment-524429459, which is to `import logging; reload(logging)`. This seems to improve the situation but not fully; I get tons of error `TypeError: unbound method emit() must be called with FileHandler instance as first argument (got RotatingFileHandler instance instead)` (and in my case this series of errors isn't great as non-technical users for my application would get alerted unnecessarily).

<details><summary>Output</summary>

```
 Traceback (most recent call last):
   File "/usr/lib/python2.7/logging/handlers.py", line 78, in emit
     logging.FileHandler.emit(self, record)
 TypeError: unbound method emit() must be called with FileHandler instance as first argument (got RotatingFileHandler instance instead)
 Logged from file core.py, line 140
 Traceback (most recent call last):
   File "/usr/lib/python2.7/logging/handlers.py", line 78, in emit
     logging.FileHandler.emit(self, record)
 TypeError: unbound method emit() must be called with FileHandler instance as first argument (got RotatingFileHandler instance instead)
 Logged from file core.py, line 140
 Traceback (most recent call last):
   File "/usr/lib/python2.7/logging/handlers.py", line 78, in emit
     logging.FileHandler.emit(self, record)
 TypeError: unbound method emit() must be called with FileHandler instance as first argument (got RotatingFileHandler instance instead)
 Logged from file core.py, line 140
```
```
# apt-cache policy ros-melodic-rospy
ros-melodic-rospy:
  Installed: 1.14.13-1bionic.20220127.143629
```
</details>

viperML, 2023-01-20:
I can't believe this is still a open issue since 2018

bhomaidan1990, 2023-02-02:
> 

NameError: name 'reload' is not defined

Lionelsy, 2024-07-05:
> @InspireX96 `NameError: name 'reload' is not defined`, in python>=3.4 solution: `importlib.reload(logging)`

A simple case / solution:
My logger is working again by adding the `importlib.reload(logging)` after `rospy.init_node`

>     rospy.init_node('ros_listener_api', anonymous=True)
>     importlib.reload(logging)

CWEzio, 2024-12-15:
Quite surprise that this bug haven't been fixed since 2018...  This bug is very annoying because it will disrupt all other module's logging behavior. It took me a while to finally realize it was `rospy` that made other module's log disappear.  I have tried 
```python
rospy.init_node('ros_listener_api', anonymous=True)
importlib.reload(logging)
```
but it did not work for me. After some work, I found a workaround:
```python
def init_ros_node(node_name):
    """
    helper function to initialize the node without disrupting logging
    """
    existing_handlers = logging.root.handlers[:]
    rospy.init_node(node_name, anonymous=True)  # this will override current logging handlers
    logging.root.handlers = existing_handlers
    # Suppress unnecessary rospy logs by setting their log level to WARNING
    rospy_logger = logging.getLogger('rospy')
    rospy_logger.setLevel(logging.WARNING)
    # Disable propagation to prevent duplicate log messages
    rosout_logger = logging.getLogger('rosout')
    rosout_logger.propagate = False
```

closed 2025-05-31 by sloretz