ros/ros_comm #1399: [rosbag] Restore snapshot functionality

ironmig, 2018-05-15:
# Background
The deprecated ```rosrecord``` command (replaced with ```rosbag record``` had an [experimental snapshot feature](http://wiki.ros.org/rosrecord#rosrecord-1) which kept the buffer in memory until triggered with a separate command to write to disk. This feature is useful in live testing scenarios where the importance of recording data is only obvious _after_ something interesting happens and is found in proprietary robotics middleware. AFAIK this feature still exists in the source, but is inaccessible from the CLI.

# Proposal
* Restore the snapshot functionality to ```rosbag record``` by exposing the option to the argument parser
* Add an option for snapshotting to buffer a fixed time duration in the past. It's more convenient for users to know they have the last 5 seconds of data like a dashcamera than thinking in terms of buffer memory usage
* Instead of triggering write with a ```std_msgs/Empty```, use an actionlib goal. This goal would allow the user to specify which topics to save, how far back to go, and name to give the file (ex: save the last 15 seconds of pointcloud data to ```cyclist_at_night.bag```). The action component would allow the user to see the progress (feedback) of writing large multigigabyte files to disk.
* Add service to pause recording, so users can have some time to think about what to save before it is overwritten

I would like to take on this project after receiving feedback on the design.

ironmig, 2018-05-15:
I found #1294 and saw that we don't want to break the existing API. Perhaps this is appropriate for a new subcommand, ```rosbag snapshot```?

ironmig, 2018-05-15:
Another option is:
* Preserve original trigger mode but also add the custom action I described
* Using --duration flag with -s flag will implement the fixed duration buffer I described
* If ```rosbag record -t``` is run, it will use the old trigger mode. However, additional flags such as ```rosbag record -t  -o filename --start -3.0 --stop -1.0 topic1 topic2 topic3...``` will use the action with the custom time, filename, and time cropping

The only API change here would be giving new purpose to options that previously did nothing

jian-dong, 2020-05-21:
@kev-the-dev  Any update about this MR?

kev-the-dev, 2020-05-21:
As discussed in the above linked PR, @DLu volunteered to make this an independent package

jian-dong, 2020-05-21:
@kev-the-dev Thanks for you reply,  this package link  [rosbag_snapshot](https://github.com/ros/rosbag_snapshot) not reachable , maybe it is private.

DLu, 2020-05-21:
Still working on it...

jian-dong, 2020-05-22:
@DLu Thank you.

DLu, 2020-06-11:
Public version availble now: https://github.com/locusrobotics/rosbag_snapshot

Working on getting it cloned at https://github.com/ros/rosbag_snapshot

dirk-thomas, 2020-06-11:
@DLu thanks for the update.

Since there is nothing to be done in this repo I will go ahead and close this ticket.

closed 2020-06-11 by dirk-thomas