ros/ros_comm #1401: Kinetic rospy uses 3x times more memory for a bare application vs indigo

v-lopez, 2018-05-16:
Running `import rospy` in a python script, requires ~40MB of memory in kinetic, in indigo it required ~10MB.

Validated on osrf/ros dockers and our robots.

Run the following code:
```
#!/usr/bin/env python
import rospy # Comment this to reduce memory usage

while True:
  pass
```

And measure with `ps faux` the RSS (resident memory)

  version | indigo | kinetic
 ------------ | ------------ | -------------
without `import rospy` | 6820 | 18728
with `import rospy` | 5724 | 49064

In our robots, our memory usage has gone up by 1GB because we have around 39 python based applications or roslaunch instances.

dirk-thomas, 2018-05-16:
Thank you for pointing out this problem. It would be great to provide more detailed information which (sub-)modules contribute to this increase.

v-lopez, 2018-05-16:
[Here](https://gist.github.com/v-lopez/94234335d2fbab1e302821fa39c7b254)'s the script I'm using for testing.

According to this, it starts with import rostime. But I still don't understand why, it might be or not related to what @jspricke says.

Here are the results of my testing:
kinetic python2.7

> python rospy_mem_test.py 
: 11288.0KB (difference 11288.0KB)
sys: 11288.0KB (difference 0.0KB)
python3: 11288.0KB (difference 0.0KB)
numbers: 11604.0KB (difference 316.0KB)
sys: 11604.0KB (difference 0.0KB)
warnings: 11604.0KB (difference 0.0KB)
rostime: 47024.0KB (difference 35420.0KB)
rospy: 51472.0KB (difference 4448.0KB)

kinetic python3
>python3 rospy_mem_test.py 
: 12316.0KB (difference 12316.0KB)
sys: 12316.0KB (difference 0.0KB)
python3: 12316.0KB (difference 0.0KB)
numbers: 12316.0KB (difference 0.0KB)
sys: 12316.0KB (difference 0.0KB)
warnings: 12316.0KB (difference 0.0KB)
rostime: 16764.0KB (difference 4448.0KB)
rospy: 26208.0KB (difference 9444.0KB)

indigo python
>: 10188.0KB (difference 10188.0KB)
sys: 10192.0KB (difference 4.0KB)
python3: 10192.0KB (difference 0.0KB)
numbers: 10220.0KB (difference 28.0KB)
sys: 10220.0KB (difference 0.0KB)
warnings: 10220.0KB (difference 0.0KB)
rostime: 11548.0KB (difference 1328.0KB)
rospy: 19684.0KB (difference 8136.0KB)

indigo python3
>: 10672.0KB (difference 10672.0KB)
sys: 10672.0KB (difference 0.0KB)
python3: 10672.0KB (difference 0.0KB)
numbers: 10672.0KB (difference 0.0KB)
sys: 10672.0KB (difference 0.0KB)
warnings: 10672.0KB (difference 0.0KB)
rostime: 12300.0KB (difference 1628.0KB)
rospy: 12300.0KB (difference 0.0KB)

jspricke, 2018-05-16:
I assume you mean this:
`from genpy.rostime import Time`
Looking into `sys.modules.keys()` it loads yaml already. Can you try to revert the patch, to test my hypothesis?

v-lopez, 2018-05-17:
Yeah looks like it's that.

The following line https://salsa.debian.org/python-team/modules/pyyaml/commit/ce6f328b074bee1e81c1f7caaf05815330a24e1e#b69ef92be68d3ff43e3437ee78d552e0458aecec_138_140

Increases the memory consumption by 31MB.

v-lopez, 2018-05-18:
I will close this issue since it's not ros_comm related, sorry for the confusion.

@jspricke Did you revert those changes or do some other kind of change?

We're going with the old regex for now.

closed 2018-05-18 by v-lopez