ros/ros_comm #1404: UDPROS does not work at lunar/armhf

AndrijCherepaha, 2018-05-20:
The issue was found during experiments with RaspberryPi. Any attempts of subscribing to topics using TransportHints.udp() or TransportHints::unreliable() end with broken communication. There is no crashes on both subscriber's and publisher's sides.

Preconditions:
Install ubuntu to armhf (https://ubuntu-mate.org/raspberry-pi/)
Install ROS Lunar as described here (http://wiki.ros.org/lunar/Installation/Ubuntu)
Compile attached example (udp) and run
  rosrun udp send
  rosrun udp recv

Expected result: recv node prints messages with received numbers.
Actual result: send prints next message
>   Error:   Received an invalid TCPROS header.  Each line must have an equals sign.
>            at line 102 in /tmp/binarydeb/ros-lunar-cpp-common-0.6.9/src/header.cpp
Found in:
Information about system:
  /rosdistro: lunar
  /rosversion: 1.13.6
Ubuntu 16.04(armhf)
ros-lunar-roscpp             1.13.6-0xenial-2018 armhf

The issue was tested using different platforms
ubuntu/armhf/lunar - the issue was found
ubuntu/armhf/jade - the issue was not reproduced
ubuntu/amd64/lunar - the issue was not reproduced
ubuntu/amd64/jade - the issue was not reproduced (upstream packages)

[udp.zip](https://github.com/ros/ros_comm/files/2020600/udp.zip)

gavanderhoorn, 2018-05-20:
Question on ROS Answers: [UDPROS communication fails at armhf](https://answers.ros.org/question/291076).

Rationale for posting here: if what @AndrijCherepaha describes is correct, this could be a regression.

pumaking, 2019-05-18:
UDP also does not work for me in Melodic. It works fine on x86_64 but I get

```
Error:   Received an invalid TCPROS header.  Each line must have an equals sign.
         at line 103 in /tmp/binarydeb/ros-melodic-cpp-common-0.6.12/src/header.cpp
```

on an Odroid XU4 (armv7l).

pumaking, 2019-09-20:
This problem is still here on Melodic.

`Error:   Received an invalid TCPROS header.  Each element must be prepended by a 4-byte length.
         at line 89 in /tmp/binarydeb/ros-melodic-cpp-common-0.6.12/src/header.cpp`

Is anyone working on this or have any tips as to where to start or what may have gone wrong?

This is core ROS functionality completely not working.

gavanderhoorn, 2019-09-21:
I would assume no one is working on this.

kjaget, 2019-10-24:
+1 on this - ROS melodic on a Jetson TX2:

rosbag record --udp <topic> 

gives 

Error:   Received an invalid TCPROS header.  Each element must be prepended by a 4-byte length.
         at line 89 in /tmp/binarydeb/ros-melodic-cpp-common-0.6.12/src/header.cpp

jveitchmichaelis, 2020-03-17:
I'm willing to look at this, as it's completely blocking embedded development. This also fails on aarch64, not just armhf.

Running the udp code above I get a different error:

```
[ERROR] [1584487855.546510052]: Header from subscriber did not have the required elements: md5sum, topic, callerid
```

I've cloned the repository and checked out `melodic-devel`. Will try and do some debugging and see what's going on. I'm not familiar with how ROS works at a low level, so I'm assuming that a C++ client will have issues with ros_comm/roscpp and a Python client will call ros_comm/rospy?

If that's the case does the issue occur with a python receiver/subscriber?

jveitchmichaelis, 2020-03-18:
This does _not_ occur with the publish/subscribe rospy examples. http://wiki.ros.org/rospy/Overview/Publishers%20and%20Subscribers

Some more experiments. I modified @AndrijCherepaha 's example to publish strings. I made four launch files to cover the possible combinations of clients.

### C++ publisher / C++ subscriber:

* `rostopic echo /chatter` prints "hello world" messages
* When launched via rosrun: `[ERROR] [1584552392.272356504]: Header from subscriber did not have the required elements: md5sum, topic, callerid`
* When launched with roslaunch: `Error:   Received an invalid TCPROS header.  Each line must have an equals sign.
         at line 103 in /tmp/binarydeb/ros-melodic-cpp-common-0.6.13/src/header.cpp`

### Python publisher / Python subscriber

* `rostopic echo /chatter` prints "hello world" messages
*  Messages are received correctly by the subscribing node

### Python publisher / C++ subscriber

* No response from subscriber (nothing printed at all, odd)

### Python subscriber / C++ publisher

* Messages are received correctly by the subscribing node

So it seems like there is perhaps a problem with the C++ subscriber?

### Code:
``` python
recv.cc

#include <ros/ros.h>
#include <std_msgs/String.h>

void udpCB (const std_msgs::StringConstPtr& str) {
  ROS_ERROR("Got: [%s]:", str->data.c_str());
}

int main(int argc, char **argv) {
  ros::init(argc, argv, "recv");
  ros::NodeHandle nh;
  ros::Subscriber sub = nh.subscribe(std::string("chatter"), 1, udpCB, ros::TransportHints().unreliable());
  ros::spin();
  return 0;
}

```

``` python
recv_py

#!/usr/bin/python

import rospy
from std_msgs.msg import String

def callback(data):
    rospy.loginfo("Got: %s", data.data)

def listener():
    rospy.init_node('subscriber')
    rospy.Subscriber("chatter", String, callback)
    rospy.spin()

if __name__ == "__main__":
    listener()

```

``` c++
send.cc

#include <ros/ros.h>
#include <std_msgs/String.h>

int main(int argc, char **argv) {
  ros::init(argc, argv, "send");
  ros::NodeHandle nh;
  ros::Publisher pub = nh.advertise<std_msgs::String>(std::string("chatter"), 1, false);
  ros::Rate r(10);

  int i = 0;
  std_msgs::String msg;
  while (ros::ok()) {

    std::stringstream ss;
    ss << "hello world (c++) " << i;
    msg.data = ss.str();

    pub.publish(msg);
    ++i;
    r.sleep();
  }

  return 0;
}

```

``` python
send_py

import rospy
from std_msgs.msg import String

pub = rospy.Publisher('chatter', String, queue_size=10)
rospy.init_node('publisher')
r = rospy.Rate(10) # 10hz

while not rospy.is_shutdown():
   pub.publish("hello world")
   r.sleep()

```

jveitchmichaelis, 2020-03-18:
Some more progress (I think):

* Overlaying `ros_comm:kinetic-devel` inside a ROS `melodic` install works as expected, no errors with unreliable transport.

* Using `melodic-devel` and checking out `kinetic-devel` specifically for `utilities/xmlrpcpp` works.

* The libxmlrpc tests in `melodic-devel` fail on ARM with `-WNarrowing`. And this is something that did change between kinetic and lunar after this commit: https://github.com/ros/ros_comm/commit/7019fb1604363c5ca1849a88930202c7ce28aed6#diff-abead072a4353c6c959717c854289c83

So I'm going to suggest this is an issue with xmlrpcpp on ARM.

jveitchmichaelis, 2020-03-18:
Ah well after all that I believe this has been fixed recently? 

https://github.com/ros/ros_comm/issues/1768

closed 2020-03-19 by dirk-thomas