ros/ros_comm #1408: ros_comm 1.4.1 breaks socketcan_bridge tests

ipa-mdl, 2018-05-23:
I just noticed that one test started to fail for melodic shadow-fixed (1.4.1), but works for main (1.3.6).
https://travis-ci.org/ipa-mdl/ros_canopen/builds/381469509

The [failing test](https://github.com/ros-industrial/ros_canopen/blob/melodic-devel/socketcan_bridge/test/to_socketcan_test.cpp#L157), runs `ros::spinOnce()` to fetch one message.

Digging through the commit https://github.com/ros/ros_comm/pull/1365 raised my attention.
I will run further tests, but this seems to change the behaviour at least for this test.

ipa-mdl, 2018-05-23:
I was not able to fix this test with an additional wait. It might be that the message got lost.
So it looks more like https://github.com/ros/ros_comm/pull/1393 is the cause.

The only external depedencies are
```
executing command [apt-get install -y ros-melodic-roscpp]
executing command [apt-get install -y ros-melodic-class-loader]
executing command [apt-get install -y ros-melodic-message-runtime]
executing command [apt-get install -y ros-melodic-std-msgs]
executing command [apt-get install -y ros-melodic-roslint]
executing command [apt-get install -y ros-melodic-rostest]
executing command [apt-get install -y ros-melodic-rosunit]
executing command [apt-get install -y ros-melodic-message-generation]
executing command [apt-get install -y libboost-all-dev]
```

dirk-thomas, 2018-05-23:
@guillaumeautran Can you please take a closer look at this.

mikepurvis, 2018-05-23:
Assuming you're on Linux, I'd be doubtful that #1393 could directly be the issue, since that was a fix for non-epoll systems such as macOS.

Does the test still fail if you build ros_comm from source but override `HAVE_EPOLL=0` in this spot?

https://github.com/ros/ros_comm/pull/1217/files#diff-af96143090deaa643970030f962c964dR70

That would kick it back to the legacy poll implementation for handling IO.

guillaumeautran, 2018-05-23:
I am taking a look right now but the specific tests passes for me at this time. Still doing some more digging.

ipa-mdl, 2018-05-23:
> Assuming you're on Linux

The test case (#1409) fails on travis, ROS buildfarm and in a local docker environment (`run_ci`).

>  Does the test still fail if you build ros_comm from source but override HAVE_EPOLL=0 in this spot?

I will push a commit to the test case PR

I did not have a deeper look

guillaumeautran, 2018-05-23:
So, looks like a setup/teardown problem. I was able to reproduce the issue (only happens on the ros_comm melodic branch) but a soon as I comment out the first test case, the second test (which was failing originally) starts to pass. 
Something must not have been cleaned up properly between the tests.

ipa-mdl, 2018-05-23:
The only other difference of package versions is `rosconsole`: 1.13.6 -> 1.13.7

ipa-mdl, 2018-05-23:
> Something must not have been cleaned up properly between the tests.

If I add a `ros::NodeHandle` to `main()`, both tests pass.

ipa-mdl, 2018-05-23:
Finally, I ran `git bisect`. The test case  started to fail with d21a33cdcee55ac09a58f702baf415539eebbacf (#1241)

dirk-thomas, 2018-05-23:
> Finally, I ran `git bisect`. The test case  started to fail with d21a33cdcee55ac09a58f702baf415539eebbacf (#1241)

@trobertson-cpr FYI

ipa-mdl, 2018-05-31:
I have resolved this by adding the `NodeHandle`.
Nevertheless, #1241 changed the behavior and affects at least the logging performance in rospy.

closed 2018-05-31 by ipa-mdl