ros/ros_comm #1411: [roslaunch] listener for dieing remote processes are not forwarded correctly

kikass13, 2018-05-24:
I have a problem with the existing roslaunch python API [kinetic].

see:
http://wiki.ros.org/roslaunch/API%20Usage
http://docs.ros.org/hydro/api/roslaunch/html/index.html

I am developing some features on top of the default roslaunch api and while playing around with some remote nodes (xmlrpc) i noticed, that there is no way to react to the death-event of a node via the existing process-manager.

I want my process monitor to be able to call my own listener object when a remote process is dieing., so [in parent.py] i added the listeners to the ParentNode constructor call:

```
def _start_server(self):
        self.logger.info("starting parent XML-RPC server")
        # OLD CALL
        #self.server = roslaunch.server.ROSLaunchParentNode(self.config, self.pm) 
        
       # NEW CALL
       self.server = roslaunch.server.ROSLaunchParentNode(self.config, self.pm, self.process_listeners)

        self.server.start()
```   
        
The implementation of 

```
class ROSLaunchParentNode(ROSLaunchNode):
    def __init__(self, rosconfig, pm):
        self.listeners = []

        handler = ROSLaunchParentHandler(pm, self.child_processes, self.listeners)
        super(ROSLaunchParentNode, self).__init__(handler)
```

does not allow the previously know listeners to be given down. 
Instead i just changed the code to 

```
class ROSLaunchParentNode(ROSLaunchNode):
    def __init__(self, rosconfig, pm, listeners=[]):
        self.listeners = listeners

        handler = ROSLaunchParentHandler(pm, self.child_processes, self.listeners)
        super(ROSLaunchParentNode, self).__init__(handler)
```
 ... now i am able to receive my callback function inside my own listener whenever a node dies remotely.

My test launchfile looks like this:

```
<launch>

  <machine name="59129943bd62" address="192.168.124.3" env-loader="/sim_sandbox/simulation_ws/devel/env.sh" user="root"/>

  <node machine="59129943bd62" name="container_detection_node1" pkg="container_detection" type="container_detection_node" output="screen" > <!-- required="true"> -->
    <rosparam command="load" file="$(find container_detection)/launch/parameters_rl.yaml" />
  </node>

  <node machine="59129943bd62" name="container_detection_node2" pkg="container_detection" type="container_detection_node" output="screen" > <!-- required="true"> -->
    <rosparam command="load" file="$(find container_detection)/launch/parameters_rl.yaml" />
  </node>

</launch>
```

after launch this was shown in my console:

> [192.168.124.3-0]: launching nodes...
> [192.168.124.3-0]: ROS_MASTER_URI=http://192.168.123.62:11311/
> [192.168.124.3-0]: process[container_detection_node1-1]: started with pid [10789]
> [192.168.124.3-0]: process[container_detection_node2-2]: started with pid [10790]
> [192.168.124.3-0]: ... done launching nodes

i then killed a node (SIGTERM) and the following output appeared

> [192.168.124.3-0]: [container_detection_node2-2] process has died [pid 10790, exit code -15, cmd /sim_sandbox/simulation_ws/devel/lib/container_detection/container_detection_node __name:=container_detection_node2 __log:=/root/.ros/log/7f413521-5e8d-11e8-aaf8-c85b76df1493/container_detection_node2-2.log].
> log file: /root/.ros/log/7f413521-5e8d-11e8-aaf8-c85b76df1493/container_detection_node2-2*.log

on top of that my callback function

```
def processDiedCallback(self, process_name, exit_code):
		print("<><> processDiedCallback <><>")
		print(process_name)
		#notify parent instance
		if(self.callback is not None):
			report = Report(sender=self)
			self.callback(report)
```

which was encapsulated inside a listener (the list of listeners i passed down into the ROSLaunchParent Object) like this

```
class MyHandler(ROSLaunchListener):
	"""
	Simple listener that receives ProcessListener events of a roslaunch server
	"""
	def __init__(self, deathCallback):
		self.deathCallback = deathCallback

	def process_died(self, process_name, exit_code):
		#we can now react to 
		self.deathCallback(process_name, exit_code)
```

and

```
self.rosLaunchParent = roslaunch_parent.ROSLaunchParent(uuid, self.args, is_core=self.options.core, 
				port=self.options.port, local_only=self.options.local_only, process_listeners = self.listeners,  #listeners will be given here
				verbose=self.options.verbose, force_screen=self.options.force_screen, is_rostest=False)
			
			self.rosLaunchParent.start()
```

outputs some random information directly after the process has died error 

> 
> <sts_overwatch.Zenyatta.MyHandler object at 0x7efdaceabbd0>
> _____________________________XAXA__________________
> <><> processDiedCallback <><>

Is this a bug? I mean it should be possible to react to dieing nodes or am i not seeing something important here?

if this is usefull i can provide a commit and a pull request for the given problem.

dirk-thomas, 2018-05-24:
This seems related to #1319. What version of the `ros_comm` package do you have installed?

kikass13, 2018-05-25:
this is indeed related, sorry i did not see that earlier. Im using ros kinetic ...

> rosversion roslaunch
> 1.12.12

dirk-thomas, 2018-05-25:
Please try the latest version then (currently 1.12.13) which includes the referenced patch. If that addresses your problem please close this ticket. Otherwise add more information what is missing. Thanks.

danielwangksu, 2019-08-16:
I think this can be closed

closed 2019-08-17 by kikass13