ros/ros_comm #1417: rosbag: allow writing message to rosbag without knowing its type (C++)

fzimmerm, 2018-06-01:
Hi,

I'm using the C++ rosbag API, and want to delete a single message from a bag. For this, I was thinking of opening the bag in `Read` mode, opening another bag in`Write` mode, and then iterate over all `MessageInstance` in a `View` of the original bag and write them to the new bag, except for messages with the same topic and type as the message I want to delete. But since the bag may contain messages of unknown types, I cannot instantiate the `MessageInstance`. I saw that `MessageInstance` has a method for writing it into a `Stream`, and that `Bag` has a private method `void Bag::write(char const* s, std::streamsize n)`, so in theory it should be possible to serialize all `MessageInstance` into a `Stream` and writing it into the new bag, without knowing the individual message types, if the `write` method weren't private.

Long story short, it would be very useful to have a method to transfer messages between bags without knowing each message's type, in C++ (I think in python this doesn't pose a problem).

ibtaylor, 2018-06-01:
`auto msg = message_instance.instantiate<topic_tools::ShapeShifter>()`
You should be able to then write that to another bag without having the c++ message type.

Unfortunately, the message instance only has when the message was written "bag write time" and not the "message time". For that, if you want to handle arbitrary types, you might want to look into something like `ros_type_introspection`.

fzimmerm, 2018-06-01:
Thanks a lot for the hint, this should solve my problem!
I'll post more feedback as soon as I have the time to try it out.

fzimmerm, 2018-06-04:
I've tried writing all messages to a new bag using `topic_tools::ShapeShifter`. The resulting bag looks fine (same size, `rosbag info` output looks correct), but when iterating over it using a `rosbag::View` it will suddenly throw a `ros::serialization::StreamOverrunException`. This isn't the case with the original bag. The same thing happened when using python to rewrite the bag.

cwecht, 2019-02-28:
@fzimmerm could you provide a minimal example of your code and a (ideally small) rosbag which reproduce this issue?

karthiknit, 2020-12-03:
@fzimmerm Could you post a sample code here for writing all messages to a bag using topic_tools::ShapeShifter?

fzimmerm, 2020-12-03:
@cwecht @karthiknit
Sorry but since I created this issue I've changed position, don't have access to the code in question anymore and am working on a topic unrelated to ROS. Thank you anyway!

shaozu, 2021-04-14:
```
for(rosbag::MessageInstance const mi : view)
{
    out_bag.write(mi.getTopic(), mi.getTime(), mi);
}
```
Just found this will do the job.

karthiknit, 2021-04-14:
I think this does the job only when you reading from one bag and writting data to other bag

shaozu, 2021-04-15:
Yes, and I guess this will meet the requirement of **fzimmerm**.

fjandrad, 2023-02-28:
I can confirm that [solution by @shaozu ](https://github.com/ros/ros_comm/issues/1417#issuecomment-819561371) . I am able to ready a bag separate topics between the ones I want the ones I dont want and then write to the new bag only the ones I desire.

closed 2025-05-31 by sloretz