ros/ros_comm #1419: Error opening too many files at once

erichahn, 2018-06-05:
tl;dr; When opening all bags in a directory if over x number rosbag crashes/fails to run.

I have a folder for all my logged tests. Inside this log folder I create individual folders for all topics. This can lead to dozens of topics with hundreds of files in each. When I try and rosbag play all the files, if I have too many, rosbag will fail to play any of them unless I merge them into fewer bags, or none at all :-P

usage case is : rosbag play /logs/*/**/*.bag --clock

example error is below.
[FATAL] [1528220861.790618714]: Error opening file: file.bag

If I delete say, half the folders (half the files) and try the same command everything runs happily. Any thoughts/ideas/questions?

jlblancoc, 2020-04-12:
Surprisingly, this still seems to be an issue. Exactly the same case as OP.
Will try to dig into the source code to find any better debug information apart of the "Error opening file" message...

jlblancoc, 2020-04-12:
Ok, after investigating ros_comm sources, I found the error had **nothing** to do with this package, but with a **system setting**: by default, each process is allowed to open up to 1024 file handles (!).

Solution: 
* `/etc/systemd/system.conf`: Uncomment and set `DefaultLimitNOFILE=65000`
* `/etc/systemd/user.conf`: Uncomment and set `DefaultLimitNOFILE=65000`
* Reboot.

Sources: [1](https://askubuntu.com/questions/1049058/how-to-increase-max-open-files-limit-on-ubuntu-18-04), [2](https://unix.stackexchange.com/questions/36841/why-is-number-of-open-files-limited-in-linux).

I guess this ticket could be closed now (cc: @dirk-thomas )

closed 2020-05-13 by dirk-thomas