ros/ros_comm #1431: roslaunch - pass through command-line args to the xmlloader when using the API in kinetic

srsidd, 2018-06-12:
#1115 enables the user to pass command-line args to the xml loader from lunar and upwards. Going through some of the discussions it seems the design was made in such a way that the feature is fully backwards-compatible and minimal changes were made to add the new functionality.

Is there a plan to back port this to Kinetic?

dirk-thomas, 2018-06-12:
In the PR #1115 you will find a referenced ticket #1323 in which all changes from `lunar-devel` for a specific time span were considered for backporting. For this patch the following is noted:

> pass through command-line args to the xmlloader when using the API #1115
> * new feature, not backported

srsidd, 2018-06-14:
I see. Is there a process defined on how to back port a feature? I could look into it if it's cherry picking the specific commits from the `lunar-dev` into the `kinetic-dev` branch.

closed 2025-05-31 by sloretz