ros/ros_comm #1433: memory leak ROSOutAppender in ros::start (checked by valgrind)

oka1125, 2018-06-14:
When I checked memory leak of my program(cpp) by using valgrind, I found memory leak of roscpp.

valgrind's report is as follows:

==6326== 304 bytes in 1 blocks are possibly lost in loss record 44 of 47
==6326==    at 0x4C2FB55: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==6326==    by 0x40138A4: allocate_dtv (dl-tls.c:322)
==6326==    by 0x40138A4: _dl_allocate_tls (dl-tls.c:539)
==6326==    by 0x741626E: allocate_stack (allocatestack.c:588)
==6326==    by 0x741626E: pthread_create@@GLIBC_2.2.5 (pthread_create.c:539)
==6326==    by 0x71F82E8: boost::thread::start_thread_noexcept() (in /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0)
==6326==    by 0x4F636B9: ros::ROSOutAppender::ROSOutAppender() (in /opt/ros/kinetic/lib/libroscpp.so)
==6326==    by 0x4F67CC4: ros::a() (in /opt/ros/kinetic/lib/libroscpp.so)
==6326==    by 0x4F349D7: ros::NodeHandle::construct(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) (in /opt/ros/kinetic/lib/libroscpp.so)
==6326==    by 0x4F34DA8: ros::NodeHandle::NodeHandle(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&) (in /opt/ros/kinetic/lib/libroscpp.so)
==6326==    by 0x4EE0E9: ToroboState::ToroboState(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, int> > >, bool) (ToroboState.cpp:27)
==6326==    by 0x4BA84E: main (torobo_dynamics_node.cpp:102)
==6326== 

According to this report, "ROSOutAppender" instance created in ros::start() function may not be released in ros::shutdown() function.
    created:
        https://github.com/ros/ros_comm/blob/kinetic-devel/clients/roscpp/src/libros/init.cpp#L348
    to be deleted:
        https://github.com/ros/ros_comm/blob/kinetic-devel/clients/roscpp/src/libros/init.cpp#L596
          (*) instance is not deleted. but assigned by value 0.

Please fix this memory leak.

tahsinkose, 2018-09-07:
I'm facing the exact problem and it seems you found the solution. You should make a pull request so that we can utilize that too.

tahsinkose, 2018-09-10:
I have fixed it in #1503. Great thanks to @oka1125 that has found the bug.

oka1125, 2018-09-11:
I'm sorry for my late reply.
And great thank you for your fixing.

tahsinkose, 2019-04-15:
Fixed in #1503.

closed 2019-04-15 by dirk-thomas