ros/ros_comm #1448: rosbag_storage - Backport isOpen() to kinetic

MasterEric, 2018-07-05:
@racko developed the convenient function for lunar and above, but this change was not backported to kinetic. Since I still have projects which require kinetic, I am stuck being unable to obtain the status of the bag file (since the bag file is private, and unable to be queried), and I have to track the bag file's status manually.

artivis, 2019-08-09:
For reference, `isOpen` was introduced in https://github.com/ros/ros_comm/pull/1190.

artivis, 2019-08-09:
This feature was marked as not being backported in https://github.com/ros/ros_comm/pull/1205, yet a backport is proposed in #1789. Up to the maintainers.

artivis, 2020-02-11:
This issue can be closed as #1789 was merged by @dirk-thomas.

PR #1789 (2020-02-10): [kinetic] Added rosbag::Bag::isOpen
Backport #1190 to kinetic-devel.

Fix #1448.
closed 2020-02-11 by dirk-thomas