ros/ros_comm #1449: ros_comm: python "message_filters.Cache.getLastestTime" should probably be "message_filters.Cache.getLatestTime"

madebyollin, 2018-07-06:
In the C++ `message_filters.Cache` API there is a method called [`getLatestTime()`](http://docs.ros.org/api/message_filters/html/c++/classmessage__filters_1_1Cache.html#a24f75357ca6deb99cac4c32c541f5013). ([source code](http://docs.ros.org/api/message_filters/html/c++/cache_8h_source.html#l00295))

But the python equivalent of this method is called [`getLastestTime()`](http://docs.ros.org/api/message_filters/html/python/#message_filters.Cache.getLastestTime). ([source code](https://github.com/ros/ros_comm/blob/e96c407c64e1c17b0dd2bb85b67f388380527097/utilities/message_filters/src/message_filters/__init__.py#L159))

This seems like a typo and is potentially confusing to people moving between the two APIs. Assuming the only things that need to be changed are the function definition and the associated unit test, I can try making a PR, but it's quite a minor change so I'm not sure if that's warranted.

dirk-thomas, 2018-07-06:
A PR against the latest development branch (currently `melodic-devel`) would certainly be appreciated. To not break backward compatibility it would be good to keep the wrongly named function, add a note to its docblock and call the new correctly spelled function in it.

madebyollin, 2018-07-06:
Added a PR [here](https://github.com/ros/ros_comm/pull/1450), although it ran into the `subPub` issue during CI testing 😢

cwecht, 2019-05-02:
This has been fixed in #1450, so this issue can be closed, right?

closed 2019-05-02 by madebyollin