ros/ros_comm #1474: Declaring a rosbag::Bag member in a nodelet class leads to failure to launch the nodelet

kartikmohta, 2018-08-06:
Having a class member of type rosbag::Bag in a nodelet leads to failure to launch the nodelet in ROS melodic.
Here is a minimum example:
```cpp
#include <nodelet/nodelet.h>
#include <rosbag/bag.h>

class MyNodelet : public nodelet::Nodelet
{
 public:
  void onInit() {}

 private:
  rosbag::Bag bag_;
};

#include <pluginlib/class_list_macros.h>
PLUGINLIB_EXPORT_CLASS(MyNodelet, nodelet::Nodelet);
```
(Here is the full package: [my_package.tar.gz](https://github.com/ros/ros_comm/files/2264378/my_package.tar.gz))

Compiling and running
```
rosrun nodelet nodelet standalone my_package/MyNodelet
```
leads to the following error:
```
[ERROR] [1533595561.186295932]: Failed to load nodelet [/my_package_MyNodelet] of type [my_package/MyNodelet] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type rosbag::NoEncryptor as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1533595561.187418380]: The error before refreshing the cache was: MultiLibraryClassLoader: Could not create object of class type rosbag::NoEncryptor as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
```
This used to work fine in lunar. It seems to be related to some recent changes to add encryption support in rosbag (https://github.com/ros/ros_comm/pull/1206).

Changing the member to be a pointer which is initialized in the `onInit` function works fine.

dseifert, 2018-08-18:
I have run into the same issue. Significant difference here is that I only create the rosbag::Bag instance after onInit was already called (local variable), actually in a callback. I have not yet found out how to fix this on my end.

jalkino, 2018-09-11:
I run into the same issue and had an extended debugging session. In the end i found the (possible) root cause - wrong setup of the plugin system in the recently introduced encryption plugins. The attached PR solved the issue for me. Happy for any feedback :-)

dseifert, 2018-09-11:
@jalkino thanks, this does indeed fix the issue.

PR #1499 (2019-03-28): fix #1474 move bag encryption plugins into separate library
Moves the bag encryption plugins into a separate library, i.e. keep it out of `librosbag_storage.so`
This setup is required by pluginlib's `class_loader`, see http://wiki.ros.org/class_loader#Caution_of_Linking_Directly_Against_Plugin_Libraries

This fixes #1474 which was caused by `librosbag_storage.so` containing directly linked code (like `rosbag:bag`) _and_ the encryption plugins. This setup is unsupported by class_loader. As described in the link, the `class_loader` tries to make it work anyway for backward compatibility by guess [... cut]
commit af0fbaac (2019-03-28): fix #1474 move bag encryption plugins into separate library (#1499)

* fix #1474 move bag encryption plugins into separate library

This setup is required by the class_loader, which is part of the pluginlib, see  http://wiki.ros.org/class_loader#Caution_of_Linking_Directly_Against_Plugin_Libraries

* fix unit tests of aes_encryptor
closed 2019-03-28 by dirk-thomas