ros/ros_comm #1487: Unexpected $(dirname) behaviour when value assigned to arg.

mikepurvis, 2018-08-16:
The existing test is here: https://github.com/ros/ros_comm/blob/e96c407c64e1c17b0dd2bb85b67f388380527097/tools/roslaunch/test/xml/test-dirname.xml

However, if the test is augmented to be:

```
<launch>
  <param name="foo" value="$(dirname)/bar" />
  <include file="$(dirname)/test-dirname/included.xml">
    <arg name="baz" value="$(dirname)/baz" />
  </include>
</launch>
```

And then the `included.xml` file is changed to:

```
<launch>
  <arg name="baz" />

  <param name="bar" value="$(dirname)/baz" />
  <param name="baz" value="$(arg baz)" />
</launch>
```

You'd expect the `baz` parameter to have `tools/roslaunch/test/xml/baz` as its path, but instead it's `tools/roslaunch/test/xml/test-dirname/baz`. Basically, the dirname is being resolved at the point at which the value is assigned to the parameter, rather than right away when the arg value is determined.

I don't have a sense how much effort this will be to fix, but I had an instance recently where it would have been nice to be able to pass a "back path" into an included file, and wasn't able to do so due to this problem.

danielwangksu, 2019-08-16:
this indeed is a issue

v4hn, 2021-06-28:
This is entirely unintuitive behavior demoting the use of the relatively new `$(dirname)`.
Sad to see this unresolved three years after Mike went through the trouble to document it...

mikepurvis, 2021-06-28:
@v4hn In fairness to the maintainers of this repo, I am in fact the author of the original `$(dirname)` patch, so this ticket was as much about self-disclosure as it was anything else. :) We ended up reworking our launchfiles to not require this fixed, but I left the ticket up in case someone else was interested in tackling it.

v4hn, 2021-06-28:
> I am in fact the author of the original `$(dirname)` patch,

Thanks! Without this bug, it would be an even more useful command, though it already helps with relative includes. :sunglasses:

> We ended up reworking our launchfiles to not require this fixed, but I left the ticket up in case someone else was interested in tackling it.

Yeah, the usual lot of bugs that are relatively easy to work around...
They just never disappear unless some good soul has the interest and time to fix it.

PR #2173 (2022-01-25): Fix $(dirname) resolution in include tags
Fixes #1487. The problem was that arg values were evaluated (non-lazily) in the context of the newly included file already.
This PR first augments the unit test and then fixes it.
closed 2022-01-25 by jacobperron