ros/ros_comm #1488: respawn parameter problem API

MiroslavKohut, 2018-08-17:
Hi 

I am starting my launchfiles using this tutorial http://wiki.ros.org/roslaunch/API%20Usage  (Part2). I have problem when I configured node parameter respawn = True in my launch file. Launch is not able to respawn node automatically when node is shutdown. I don't think this is normal behavior.

Thank you for your help.

dirk-thomas, 2018-08-17:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2018-08-17 by dirk-thomas

MiroslavKohut, 2018-09-04:
Ok thank you for information and I am sorry for not  following the guidelines. I already created the question. https://answers.ros.org/question/300988/respawn-parameter-problem/  but I do not think anyone would be able to help me in this specific problem.