ros/ros_comm #1491: How to launch a node with it's dependencies have started

itfanr, 2018-08-21:
I have 5 launch file. Some node must start when some nodes have started , or they cannot work.

If all the nodes are written by myself, it is easy to deal with.

But now, I use some packages like moveit, whose code is not in my workspace and I donnot want to rewrite it :

```
  <node name="move_group" launch-prefix="$(arg launch_prefix)" pkg="moveit_ros_move_group" type="move_group" respawn="false" output="screen" args="$(arg command_args)">

```

Can anyone help me?

dirk-thomas, 2018-08-21:
I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2018-08-21 by dirk-thomas

itfanr, 2018-08-21:
@dirk-thomas 
Sorry for that. and thank you for your advice.
I asked a question here:
https://answers.ros.org/question/301136/how-to-launch-a-node-with-its-dependencies-have-started/