ros/ros_comm #1492: service::waitForService stuck in ros::Duration::sleep when use_sim_time true without /clock publisher, because using ros::Duration::sleep instead of ros::WallDuration::sleep

alrros, 2018-08-22:
The function service::waitForService in file libros/service.cpp 
uses ros::Duration::sleep instead of ros::WallDuration::sleep,
which leads to the side effect of beeing stuck in ros::Duration::sleep 
under the following conditions:
- use_sim_time is set as ros parameter 
- no simulated time publisher is active (no /clock publisher)
- the service provider is not yet available

This case is significantly critical, when:
- using a nodelet as /clock publisher
- starting nodelet manager and nodelet loader in 1 roslaunch
--> in 99/100 cases everything is fine
--> in  1/100 cases the nodelet loader gets stuck in service::waitForService -> ros::Duration::sleep

Although nodelet is used as an example, the main problem is the usage of ros::Duration::sleep at service client initialization instead of ros::WallDuration::sleep.

Quick Reproduction:
The problem can be quickly reproduced the following way by using the service "load_nodelet" of the nodelet manager as an example:
case A) everything is fine
- first: terminal 1: roscore
- then: terminal 2: rosparam set use_sim_time "true"
- then: terminal 3: rosrun nodelet nodelet manager __name:=nodelet_manager
- then: (wait 1s)
- then: terminal 4: rosrun nodelet nodelet load nodelet_tutorial_math/Plus nodelet_manager __name:=nodelet1 nodelet1/in:=foo _value:=1.1
- then: terminal 5: rosservice call /nodelet_manager/list
==> nodelet is loaded

case B) stuck, no nodelet loaded
- first: terminal 1: roscore
- then: terminal 2: rosparam set use_sim_time "true"
- then: terminal 3: rosrun nodelet nodelet load nodelet_tutorial_math/Plus nodelet_manager __name:=nodelet1 nodelet1/in:=foo _value:=1.1
- then: (wait 1s)
- then: terminal 4: rosrun nodelet nodelet manager __name:=nodelet_manager
- then: terminal 5: rosservice call /nodelet_manager/list
==> no nodelet loaded, empty list

(note: no /clock publisher is active, but ros::Duration::sleep is never executed in case A)

------
------

function
- bool service::waitForService(const std::string& service_name, ros::Duration timeout) 

in file 
- ros_comm/clients/roscpp/src/libros/service.cpp

uses
- ros::Duration::sleep

instead of
- ros::WallDuration::sleep

under the following conditions
- use_sim_time is set as ros parameter 
- no simulated time publisher is active (no /clock publisher)
- the service provider is not yet available

causing critical side effect:
- beeing stuck in service::waitForService -> ros::Duration::sleep

dirk-thomas, 2018-08-22:
Please consider to provide a PR with the proposed change.

khssnv, 2019-01-07:
Facing the same in ROS Kinetic.

cwecht, 2019-02-28:
@khssnv your problem seems to have a different cause. rospys wait_for_service function relies not on rostime (as far as I can see).

PR #1638 (2019-02-28): roscpp/service: use WallTime/WallDuration for waiting
Fixes #1492
closed 2019-02-28 by dirk-thomas