ros/ros_comm #1500: ROS service with no/empty response is not supported by rospy client

yli-cpr, 2018-09-06:
https://github.com/ros/ros_comm/blob/53d958fdccb15c2c5d7ed3cef33850e2f472f7d4/clients/rospy/src/rospy/impl/tcpros_service.py#L196

Why does it raise exception here?

dirk-thomas, 2018-09-06:
Because a service handler is supposed to return the `response`.

---

I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

closed 2018-09-06 by dirk-thomas

yli-cpr, 2018-09-06:
I probably referred to the wrong place of the code. But what I have seen is ROS srv with no response data didn't work when client makes calls. And the code I referred above seems on the server side. Sorry for that.

But if I see calling a ROS srv with no response data fails again, can I create an issue here?

yli-cpr, 2018-09-06:
Actually the code I should post is here:

```
            responses = transport.receive_once()
            if len(responses) == 0:
                raise ServiceException("service [%s] returned no response"%self.resolved_name)
```

yli-cpr, 2018-09-06:
Never mind. I can't reproduce this any more. I met a problem like this a few months ago, and thought empty response service was just not supported until today. That's why I came here. But maybe something was wrong when I saw the problem, or something was fixed. Sorry for that.