ros/ros_comm #1514: rosservice find / unable to communicate with service

mistoll, 2018-09-28:
When a crashed node leaves dead service in the roscore, then rosservice find won't work any more.

Example:
```
rosservice find std_srvs/Trigger
ERROR: Unable to communicate with service [/mynode/get_loggers], address [rosrpc://localhost:52811]
```

IMO this is not a desired behavior, because the services can be completely unrelated.

This could simply be addressed by catching the errro at https://github.com/ros/ros_comm/blob/89ca9a7bf288b09d033a1ab6966ef8cfe39e1002/tools/rosservice/src/rosservice/__init__.py#L325 I can provide a pull request.

dirk-thomas, 2018-10-08:
> I can provide a pull request.

That would be great. Please also include steps to reproduce the problem.

mistoll, 2018-10-09:
Steps to reproduce:

1. Start roscore
2. Start a simple node e.g.
```
import rospy
rospy.init_node("crash_me")
rospy.spin()
```
3. Kill that
`kill -SIGKILL $!`
4. Try to use rosservice find
`rosservice find std_srvs/SetBool`

mistoll, 2019-02-04:
Closed by #1519

closed 2019-02-04 by mistoll