ros/ros_comm #1516: rosnode ping -a takes close to 30 seconds to finish

msadowski, 2018-10-03:
I'm running some stress testing of my setup that uses [node_alive], which in turn uses rosnode ping.

My setup:
1. Laptop running master
2. 3 single board PCs 
3. Network switch to link all of the above

I noticed that if I forcibly disconnect network on the single board computer by unplugging the ethernet the command node ping -a can take close to 30 seconds to finish with the following output:

```
Will ping the following nodes: 
 * /node1
 * /node2
 * /rosout
 * /node3

pinging /node1 with a timeout of 3.0s
xmlrpc reply from http://10.42.0.121:59371/     time=2.152920ms
pinging /node2 with a timeout of 3.0s
xmlrpc reply from http://10.42.0.134:46530/     time=2.267838ms
pinging /rosout with a timeout of 3.0s
xmlrpc reply from http://10.42.0.1:33253/       time=0.365973ms
pinging /node3 with a timeout of 3.0s
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
connection to [/node3] timed out
```

It looks as if the timeout is not respected?

msadowski, 2018-10-11:
Closing this since the patch was merged. Thanks!

closed 2018-10-11 by msadowski

akifh, 2019-12-15:
I think we should re-open this. Even with this patch, `ValueError` is still thrown with "timed out" string. Moreover, it seems like it takes forever with this exception:

```
$ rosnode ping -a
Will ping the following nodes: 
 * /node1
 * /node2

pinging /node1 with a timeout of 3.0s
xmlrpc reply from http://192.168.0.232:36621/	time=0.344992ms
pinging /node2 with a timeout of 3.0s
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
unknown network error contacting node: timed out
```

Unfortunately, the answer to this question by @dirk-thomas : https://github.com/ros/ros_comm/pull/1517#discussion_r223420156 is "it seems like yes".

Shall we add `return False` to `ValueError` to fix the issue?

Note: I raised the exception by calling `kill` with pid of node2. Of course, using `melodic` from apt.